moonlight-stream/moonlight-embedded #549: Raspbian (debian 9 with noobs install) installation failed (broken package)

Tsuna77, 2017-08-28:
Please provide the following info.

**Moonlight Embedded version:** _Latest in jessie repo (strectch repo not available)_
[image]

**Moonlight Embedded source:** _http://archive.itimmer.nl/raspbian/moonlight_
**Moonlight Embedded running on:** _Raspberry Pi 3 type B_
**Moonlight Embedded running on distribution:** _Raspbian_
[image]

**Verbose output `-verbose` of Moonlight Embedded:** Nothing

**What is the expected result?**
Install successfull

**What happens instead of that?**
Could not install, broken libssl1.0.0 package not available (libssl1.0.2 & libssl1.1 installed)
[image]
[image]

MatWan67, 2017-08-28:
I m experiencing the same issue

kittenslut, 2017-08-30:
Same issue, sadly.

Kohtachan, 2017-08-31:
Same Problem with raspbian stretch.

Unorthodactyl, 2017-09-01:
You can download and install libssl1.0.0 from here https://packages.debian.org/jessie/armhf/libssl1.0.0/download
this will let you install moonlight-embedded as per the wiki's package instructions
then you will have to use sudo to pair and stream.

Unorthodactyl, 2017-09-01:
Maybe don't close this issue so others can see this, also this is a workaround and the issue is still with the project.

cgutman, 2017-09-01:
It should be pretty simple for @irtimmer to compile a package for Stretch. Now that Stretch is using OpenSSL 1.1 like Arch, master should build stock without needing a special branch for the ancient OpenSSL in Jessie.

benwilesmusic, 2017-09-29:
I posted a solution here: #568

irtimmer, 2017-11-07:
Since version 2.4.4 a package for Raspbian Stretch is provided

closed 2017-11-07 by irtimmer