moonlight-stream/moonlight-embedded #550: L/R triggers not consistently recognised on Logitech Cordless RumblePad 2

Meems33, 2017-09-02:
I'm using the Logitech Cordless RumblePad 2 with moonlight streaming. It seems that in some games the L/R triggers (buttons labeled 7/8 mapped to b6/b7) are not recognized. I have used jstest to verify that the mapping in [gamecontrollerdb.txt](https://github.com/irtimmer/moonlight-embedded/blob/master/gamecontrollerdb.txt) is correct.

It is possible this is not a moonlight issue.

Here are 3 samples in Steam:
- Portal 2 - L/R triggers work
- Rocket League - L/R triggers are not recognized. When trying to configure a controller they do not work. All other buttons seem to work
- Lego Marvel Superheros - L/R triggers are not recognized. Configuration does not respond when pushing them. Other buttons seem to work.

**_NVidia Geforce Experience version:_**
3.9.0.61
**Moonlight Embedded version:**
2.4.2ish

**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
compiled against commit [3568934860bf5b3989e0e9cc79d1e577d51ac56c](https://github.com/irtimmer/moonlight-embedded/commit/3568934860bf5b3989e0e9cc79d1e577d51ac56c)

**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
Pi 3b
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._
Raspbian

> $ sudo apt list --installed / grep xpad
> xpad/oldstable,now 4.1-1 armhf [installed]

**Verbose output `-verbose` of Moonlight Embedded:**

>  moonlight -verbose stream
> Moonlight Embedded 2.4.1 (EMBEDDED;ALSA;PULSE;PI)
> Searching for server...
> Connect to 192.168.0.115...
> NVIDIA GeForce GTX 1060 6GB, GFE 3.9.0.61 (gs_04_00_22663502, 7.1.400.0)
> Platform Raspberry Pi (Broadcom)
> Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
> No mapping available for   USB Keyboard (03000000d90400000316000010010000) on /dev/input/event1
> Detected Logitech Logitech Cordless RumblePad 2 (030000006d04000019c2000010010000) on /dev/input/event4 as Logitech Cordless RumblePad 2
> Stream 1280 x 720, 60 fps, 10000 kbps
> Initializing platform...done
> Resolving host name...done
> Starting RTSP handshake...done
> Initializing control stream...done
> Initializing video stream...done
> Initializing audio stream...done
> Initializing input stream...done
> Starting control stream...done
> Starting video stream...done
> Starting audio stream...done
> Starting input stream...done
> Returning RTP packet queued for too long
> Received OOS audio data (expected 12023, but got 12024)

**What is the expected result?**
Games recognize the L/R triggers

**What happens instead of that?**
Some games are not recognizing the L/R triggers

Meems33, 2017-09-02:
I believe it would work if I could get the host pc to recognize the controller as "Logitech Logitech Cordless RumplePad 2 (Vendor: 046d Product: c219)" instead of "Xbox 360 Controller (XInput STANDARD GAMEPAD)". The latter causes the trigger buttons to register only if I press another button with it. If I press a trigger button all by itself it does not register the button push.

Meems33, 2017-09-02:
Using wireshark I can tell when a packet is received that has some kind of input change. When idle the UDP packets are of size 60 and 86. When input data comes in the size is 108 (aka a button is pressed or a stick is moved). There are no 108 byte packages received when I push the L/R triggers by themselves. Hence the issue is that pressing the trigger does not seem to be triggering something to send out a new data packets.

Meems33, 2017-09-03:
I believe I have found the problem. Looking at the code in evdev.c the trigger cases do not appear to set the gamepadModified flag to true. PR @ #552

irtimmer, 2017-09-03:
Bug have been fixed with commit a4e19af04f56a266b7d8bd657e3c3ba3210b21cc

closed 2017-09-03 by irtimmer