moonlight-stream/moonlight-embedded #553: build error: for loop initial declaration used outside C99 mode

MrCapone, 2017-09-06:
Udoobuntu based on Uboontu 14.04
gcc version: gcc (Ubuntu/Linaro 4.8.4-2ubuntu1~14.04.3) 4.8.4
cmake version 3.2.2

closed 2017-09-06 by irtimmer

MrCapone, 2017-09-06:
Update issue info as required.

irtimmer, 2017-09-06:
Fixed with commit d32b5b656fd481fe73bd300bbfd6fd411f5a612f

MrCapone, 2017-09-07:
After d32b5b6 issue still appear 
```
/home/udooer/moonlight-embedded/third_party/moonlight-common-c/reedsolomon/rs.c: In function ‘reed_solomon_new’:
/home/udooer/moonlight-embedded/third_party/moonlight-common-c/reedsolomon/rs.c:399:9: error: ‘for’ loop initial declarations are only allowed in C99 mode
         for (int row = 0; row < rs->shards; row++) {
         ^
/home/udooer/moonlight-embedded/third_party/moonlight-common-c/reedsolomon/rs.c:399:9: note: use option -std=c99 or -std=gnu99 to compile your code
/home/udooer/moonlight-embedded/third_party/moonlight-common-c/reedsolomon/rs.c:400:13: error: ‘for’ loop initial declarations are only allowed in C99 mode
             for (int col = 0; col < data_shards; col++)
             ^
/home/udooer/moonlight-embedded/third_party/moonlight-common-c/reedsolomon/rs.c:419:9: error: ‘for’ loop initial declarations are only allowed in C99 mode
         for (int j = 0; j < parity_shards; j++) {
         ^
/home/udooer/moonlight-embedded/third_party/moonlight-common-c/reedsolomon/rs.c:420:13: error: ‘for’ loop initial declarations are only allowed in C99 mode
             for (int i = 0; i < data_shards; i++)
             ^
```

```
udooer@udoo:~/moonlight-embedded$ git log
commit d32b5b656fd481fe73bd300bbfd6fd411f5a612f
Author: Iwan Timmer <irtimmer@gmail.com>
Date:   Wed Sep 6 22:14:22 2017 +0200

    Set C99 usage globally instead of per target
```

irtimmer, 2017-09-07:
Commit 06c3c7a should fix the issue with moonlight-common-c

MrCapone, 2017-09-07:
Build works without errors on gcc-5 `(Ubuntu/Linaro 5.4.1-2ubuntu1~14.04)`.
But if i switch to gcc-4 `(Ubuntu/Linaro 4.8.5-2ubuntu1~14.04.1)` i got build errors:
```
[ 41%] Building C object libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/reedsolomon/rs.c.o
/home/udooer/moonlight-embedded/third_party/moonlight-common-c/reedsolomon/rs.c: In function ‘reed_solomon_new’:
/home/udooer/moonlight-embedded/third_party/moonlight-common-c/reedsolomon/rs.c:399:9: error: ‘for’ loop initial declarations are only allowed in C99 mode
         for (int row = 0; row < rs->shards; row++) {
         ^
/home/udooer/moonlight-embedded/third_party/moonlight-common-c/reedsolomon/rs.c:399:9: note: use option -std=c99 or -std=gnu99 to compile your code
/home/udooer/moonlight-embedded/third_party/moonlight-common-c/reedsolomon/rs.c:400:13: error: ‘for’ loop initial declarations are only allowed in C99 mode
             for (int col = 0; col < data_shards; col++)
             ^
/home/udooer/moonlight-embedded/third_party/moonlight-common-c/reedsolomon/rs.c:419:9: error: ‘for’ loop initial declarations are only allowed in C99 mode
         for (int j = 0; j < parity_shards; j++) {
         ^
/home/udooer/moonlight-embedded/third_party/moonlight-common-c/reedsolomon/rs.c:420:13: error: ‘for’ loop initial declarations are only allowed in C99 mode
             for (int i = 0; i < data_shards; i++)
             ^
make[2]: *** [libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/reedsolomon/rs.c.o] Error 1
make[1]: *** [libgamestream/CMakeFiles/moonlight-common.dir/all] Error 2
make: *** [all] Error 2
```