moonlight-stream/moonlight-embedded #555: Failed to pair to server

awzbert, 2017-09-06:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.9.0.61
**Moonlight Embedded version:** 2.4.3-1jessie1
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._ http://archive.itimmer.nl/raspbian/moonlight jessie main
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._ Raspberry pi 3
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._ Raspbian Jessie

**Verbose output `-verbose` of Moonlight Embedded:** pairing to local computer, code pops i enter it and hit accept then moonlight responds with Failed to pair to server: Failed to sign data.

it worked once then i unpaired it because it was not finding any games just saying searching for server.

i have uninstalled all of my software that says it is incompatible. 

**What is the expected result?**?

**What happens instead of that?**?

shadowzero26, 2017-09-07:
I am having the same problem with raspbian stretch.  Someone else had the same problem but didn't list the solution.

shadowzero26, 2017-09-07:
awzbert, You can install the missing dependency ver 1.0.0 by downloading it separately and right clicking and installing it, or load the raspbian jessie distribution link into your update sources and apt-get it that way.  I don't recommend the latter because then you need to remove jessie from the list afterward.

It worked for me up until the latest geforce experience update ver 3.9.

awzbert, 2017-09-07:
I have moonlight installed and it runs. But it won’t pair with GeForce. I’ll leave it I till we can find a solution

shadowzero26, 2017-09-08:
Last night I used noobs to fresh install raspbian.  The only version available is currently stretch.  I had a jessie version on standby.

I loaded the dependency, then added the source for moonlight, then installed moonlight-embedded, then ran 
sudo moonlight pair
it generated a certificate then waited for a couple seconds and gave me the code to pair.  I entered it and it paired.

MAYBE, I was always typing in the whole ip every time before and that's why it never generated a certificate?  I haven't tried this on other installs but will when I get home from work.  so try
sudo moonlight pair
and see if it works without the ip.

awzbert, 2017-09-08:
Will try thanks

shadowzero26, 2017-09-09:
No, it didn't work.  I tried it on the other installation.  It could be that noobs loaded up older drivers.

Note:  I did a fresh download of noobs, then downloaded raspbian stretch, and DID NOT do sudo apt-get upgrade after I did the update to nab moonlight.

awzbert, 2017-09-10:
Mine says missing libssl1.0.0

shadowzero26, 2017-09-10:
https://www.dropbox.com/s/9w40vgxgxh6apjy/libssl1.0.0_1.0.1t-1%2Bdeb8u6_armhf.deb?dl=0
Here you go.   Just right click install it after downloading, or do some CLI stuff.

awzbert, 2017-09-10:
I have just did a full re install and it is now working with that ssl package

https://www.dropbox.com/s/9w40vgxgxh6apjy/libssl1.0.0_1.0.1t-1%2Bdeb8u6_armhf.deb?dl=0 [https://www.dropbox.com/s/9w40vgxgxh6apjy/libssl1.0.0_1.0.1t-1%2Bdeb8u6_armhf.deb?dl=0]

shadowzero26, 2017-09-11:
Glad to hear.  I think something in the recent updates on raspbian must be doing it, or a combination of that and the new GeForce experience update

rodobodolfo, 2017-10-14:
I had the same issue, and fixed by manually specifying the key directory. I think now its has to be executed with sudo the keys are elsewhere and I couldn't flush them...

sudo moonlight pair 192.168.0.99 -keydir /home/pi/.cache/moonlight/

shadowzero26, 2017-11-01:
Hmm... I'll give that a shot tonight

shadowzero26, 2017-11-25:
So, it seems to work with a fresh install now. I have NOT had to specify the key directory since I flashed the latest raspian.  I have not updated anything since 10-31-2017 and it still works great.

My one qualm is now my PS3 controllers sometimes load as PS3 controllers during a stream and other times load up as Nvidia Sheild device, which registers input in Steam and anything run directly through Big Picture mode, but if I exit or do not start Big Picture mode it is difficult to play many games because many will not recognize the Nvidia Sheild as a legitimate controller.  I've tried x360ce.exe which recognizes all of the buttons but for some strange reason will not allow me to apply the settings. I.e. when I load x360ce.exe, it asks me if I want to search internet for settings, if I say no, I cannot use it, if I say yes, it loads up a profile that allows me to see every button being pressed but if I try to use it or apply or change any settings it clears every setting instantly.  

On my Logitech bluetooth keyboard with trackpad I do have another qualm, I can't figure out a way to get the window key to map when streaming, and also any clicking and dragging I do messes up.  I'm assuming streaming the trackpad sends some awkward mixed signal that doesn't get sent over stream, because sometimes it will cause a scroll, right click, left click, or double click and drag.

posti85, 2018-01-24:
Just using `sudo` worked for me. It is strange, it stopped working suddenly...