moonlight-stream/moonlight-embedded #556: Segfault running as a service

Meems33, 2017-09-07:
I am running a service that listens to pi gpio interrupts from button pushes and performs operations in python. When trying to launch moonlight through a systemd service it segfaults. I have attached a log running with strace.

Based on the strace I suspect it segfaults here: https://github.com/irtimmer/moonlight-embedded/blob/master/src/config.c#L84

I believe that since I am running as a service the HOME variable is not set. Hence home_dir will be null and strlen(null) segfaults.

`const char *home_dir = getenv("HOME");`

Snippet of gpio python code
```
button_moonlight = 11
button_moonlight_status = 12

def signal_handler(signal, fame):
  print("cleanup")
  gpio.cleanup()
  sys.exit(0)

def loop():
  moonlightRunning = "moonlight" in (p.name() for p in psutil.process_iter())
  gpio.output(12, moonlightRunning)
  threading.Timer(2, loop).start()

def moonlight(pin):
  print("moonlight")
  tee.system(["strace", "moonlight", "-verbose", "stream"], cwd="/tmp", logger="/var/log/moonlight_gpio.log")

signal.signal(signal.SIGINT, signal_handler)

gpio.setmode(gpio.BOARD)
gpio.setup(button_moonlight, gpio.IN) # moonlight streaming button
gpio.setup(button_moonlight_status, gpio.OUT) # moonlight status LED

gpio.add_event_detect(button_moonlight, gpio.RISING, callback=moonlight, bouncetime=200)

loop()
```

**_NVidia Geforce Experience version:_**
3.9.0.61
**Moonlight Embedded version:**
2.4.3 HEAD

**Moonlight Embedded running on:**
Pi 3b
**Moonlight Embedded running on distribution:**
Raspbian

strace log from moonlight -verbose stream:

> Sep 06 23:55:55 raspberrypi python3[1632]: rt_sigaction(SIGRTMIN, {0x76a1d434, [], SA_RESTORER|SA_SIGINFO, 0x769061a0}, NULL, 8) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: rt_sigaction(SIGRT_1, {0x76a1d2d8, [], SA_RESTORER|SA_RESTART|SA_SIGINFO, 0x769061a0}, NULL, 8) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: rt_sigprocmask(SIG_UNBLOCK, [RTMIN RT_1], NULL, 8) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: getrlimit(RLIMIT_STACK, {rlim_cur=8192*1024, rlim_max=RLIM_INFINITY}) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: brk(0)                                  = 0x1f52000
> Sep 06 23:55:55 raspberrypi python3[1632]: brk(0x1f73000)                          = 0x1f73000
> Sep 06 23:55:55 raspberrypi python3[1632]: futex(0x75e08c44, FUTEX_WAKE_PRIVATE, 2147483647) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: open("/dev/urandom", O_RDONLY|O_LARGEFILE) = 3
> Sep 06 23:55:55 raspberrypi python3[1632]: fcntl64(3, F_GETFD)                     = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: fstat64(3, {st_mode=S_IFCHR|0666, st_rdev=makedev(1, 9), ...}) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: clock_gettime(CLOCK_REALTIME, {1504742155, 121728112}) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: getrusage(0x1 /* RUSAGE_??? */, {ru_utime={0, 40000}, ru_stime={0, 10000}, ...}) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: read(3, "{\241\360\316\262\350\4\274q\363\270\3312\366\342\3\"KR\26\356+\317N\323\1\344\264x\f\256\t", 32) = 32
> Sep 06 23:55:55 raspberrypi python3[1632]: read(3, "\337\256\217:d:\323\253<l-r~g\31\332[\234\3419\346\234\334\231$\\d\ts\244\307\24", 32) = 32
> Sep 06 23:55:55 raspberrypi python3[1632]: read(3, ";\272\346\2458\37\35\317", 8)  = 8
> Sep 06 23:55:55 raspberrypi python3[1632]: rt_sigprocmask(SIG_SETMASK, ~[ILL TRAP BUS FPE SEGV RTMIN RT_1], [], 8) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: rt_sigaction(SIGILL, {0x76b06025, ~[ILL TRAP BUS FPE SEGV RTMIN RT_1], SA_RESTORER, 0x76906180}, {SIG_DFL, [], 0}, 8) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: rt_sigprocmask(SIG_BLOCK, NULL, ~[ILL TRAP BUS FPE KILL SEGV STOP RTMIN RT_1], 8) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: rt_sigaction(SIGILL, {SIG_DFL, [], SA_RESTORER, 0x76906180}, NULL, 8) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
> Sep 06 23:55:55 raspberrypi python3[1632]: access("gamecontrollerdb.txt", R_OK)    = -1 ENOENT (No such file or directory)
> Sep 06 23:55:55 raspberrypi python3[1632]: --- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0} ---
> Sep 06 23:55:55 raspberrypi python3[1632]: +++ killed by SIGSEGV +++
> Sep 06 23:55:55 raspberrypi python3[1632]: Segmentation fault

Meems33, 2017-09-07:
As a temporary workaround I set the HOME environment variable in the service config and it gets past the seg fault and seems to work.

> $ cat /lib/systemd/system/gpio1.service
> [Unit]
> Description=GPIO1 Service
> After=multi-user.target
> 
> [Service]
> Type=simple
> Environment="HOME=/home/pi"
> ExecStart=/usr/bin/python3 /usr/local/bin/gpio1.py
> 
> [Install]
> WantedBy=multi-user.target

irtimmer, 2017-09-07:
With commit 82b956a the home directory should be correctly retrieved in case the HOME environment variable isn't set. In your case this would probably be /root (instead of your currently set /home/pi) as you're trying to run Moonlight as a system service.

closed 2017-09-07 by irtimmer