moonlight-stream/moonlight-embedded #561: Xbox 360 Wireless No longer recognized

sehackma, 2017-09-17:
**NVidia Geforce Experience version:3.9.0.61**
**Moonlight Embedded version:2.4.3**
**Moonlight Embedded source: current binary http://archive.itimmer.nl/raspbian/moonlight/moonlight-embedded_2.4.0-1jessie1_armhf.deb**
**Moonlight Embedded running on: Raspberry Pi 3 Model B
** 
**Moonlight Embedded running on distribution:Raspbian** 

**Verbose output `-verbose` of Moonlight Embedded:
> Moonlight Embedded 2.4.3 (EMBEDDED;ALSA;PI)
> Searching for server...
> Connect to 192.168.0.19...
> NVIDIA GeForce GTX 670, GFE 3.9.0.61 (gs_04_00_22663502, 7.1.400.0)
> Platform Raspberry Pi (Broadcom)
> Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
> Stream 1280 x 720, 60 fps, 10000 kbps
> Initializing platform...done
> Resolving host name...done
> Starting RTSP handshake...done
> Initializing control stream...done
> Initializing video stream...done
> Initializing audio stream...done
> Initializing input stream...done
> Starting control stream...done
> Starting video stream...done
> Starting audio stream...done
> Starting input stream...done
> Returning RTP packet queued for too long
> Received OOS audio data (expected 57162, but got 57163)
> Unrecoverable frame 2: 13+0=13 received < 24 needed
> Unrecoverable frame 3: 8+0=8 received < 24 needed
> Network dropped an entire frame
> Waiting for IDR frame
> IDR frame request sent
> Waiting for IDR frame
> Waiting for IDR frame
> Waiting for IDR frame
//quit using mouse & keyboard on host
> Control stream received disconnect event
> Loss Stats: Transaction failed: 11
> Stopping input stream...done
> Stopping audio stream...done
> Stopping video stream...done
> Stopping control stream...done
> Cleaning up input stream...done
> Cleaning up audio stream...done
> Cleaning up video stream...done
> Cleaning up control stream...done
> Cleaning up platform...done

**

**What is the expected result?**
The controls are set up using the standard /usr/share/moonlight/gamecontrollerdb.txt, have tried multiple fixes from previous support tickets, the config seems to match my jstest / evtest

**What happens instead of that?**
The big picture in steam recognizes no controllers connected. This all began when updating from I believe 2.3.0, not certain on previous version but was working successfully with custom "xbox.map" I created using "moonlight map"

sehackma, 2017-09-22:
I reverted to 2.3.0-1jessie1 and used map as previously and the problem remains. Note that the controllers sync properly with emulation station, retrospect, etc. and are recognized using "Moonlight map" just not recognized by steam big picture when streaming.

rasluht, 2017-10-12:
Hey there, I'm having a similar problem. Everything worked fine a few weeks ago. But somthing has been updated and now I cant play CEMU over Moonlight (Raspberry Pi3) anymore. The 360 controller isn't beeing recognized anymore. In Steam Big Picture everything seems OK and works. Bu I can only configure 1 controller and the binding parts show me chinese symbols. Normally it showed me XBOX 360 Controller in the dropdown now it says "NIVIDIA SHIELD". Anybody have a solution for me?

1031982, 2017-11-11:
I have the same basic issue as well. RetroPie sees the controller just fine, connecting the receiver to my PC causes the controller to work perfectly there.
As soon as I try Moonlight, it either doesn't work at all or half of it will.
I have an official receiver, and have now re-installed everything on my Raspberry Pi.
From a clean install it has the same issue.

dhd0004, 2018-04-03:
Hi, i have the same issue with Raspberry Pi 3b+ and last version of RetroPie.
Did you found any fix?

regards

sehackma, 2018-04-07:
I did not but the latest version maps closer to correct and seems to be reading gamecontrollerdb. I haven't tried manually remapping as I'm hoping irtrimmer fully fixes the re-added map function in the next release.

closed 2018-04-07 by sehackma