moonlight-stream/moonlight-embedded #563: Controller with GUID in gamecontrollerdb.txt mapped differently

StrayDog16, 2017-09-22:
NVidia Geforce Experience version:3.9.0.61
Moonlight Embedded version:2.4.3
Moonlight Embedded source: current binary http://archive.itimmer.nl/raspbian/moonlight/moonlight-embedded_2.4.0-1jessie1_armhf.deb
Moonlight Embedded running on: Raspberry Pi 3 Model B
Moonlight Embedded running on distribution:Raspbian

This may have a simple enough answer but I have an Xbox One Controller (Firmware 2015) GUID 030000005e040000dd02000003020000 that is currently in gamecontrollerdb.txt however it's mapped differently. 

Instead of the published 030000005e040000dd02000003020000,Microsoft X-Box One pad v2,x:b2,a:b0,b:b1,y:b3,back:b6,guide:b8,start:b7,dpleft:h0.8,dpdown:h0.0,dpdown:h0.4,dpright:h0.0,dpright:h0.2,dpup:h0.0,dpup:h0.1,leftshoulder:h0.0,leftshoulder:b4,lefttrigger:a2,rightshoulder:b5,righttrigger:a5,leftstick:b9,rightstick:b10,leftx:a0,lefty:a1,rightx:a3,righty:a4,platform:Linux,

This worked for me 030000005e040000dd02000003020000,Xbox One Controller (Firmware 2015),x:b2,a:b0,b:b1,y:b3,back:b8,guide:b10,start:b9,dpleft:h0.8,dpdown:h0.0,dpdown:h0.4,dpright:h0.0,dpright:h0.2,dpup:h0.0,dpup:h0.1,leftshoulder:h0.0,leftshoulder:b4,lefttrigger:b7,rightshoulder:b5,righttrigger:b6,leftstick:b11,rightstick:b12,leftx:a0,lefty:a1,rightx:a2,righty:a3,platform:Linux,

Am I missing something as to why it'd be mapped differently?

Thanks!

irtimmer, 2017-09-22:
The Xbox One controller has analog triggers and shomehow you have misconfigured your gamepad to only let it show up as digital triggers. You can see this difference in the lefttrigger and righttrigger options. Because this result in a slightly different order of the buttons the leftstick and rightstick buttons also no longer work with the default mapping.

closed 2017-09-22 by irtimmer

StrayDog16, 2017-09-22:
I forgot to specify that I'm running this through RetroPie which is where I assume the improper configuration is happening.

I'm also using controller 030000005e040000ea02000001030000,Microsoft X-Box One pad v2 and it's mapping in gamecontrollerdb.txt works fine with analog triggers. So not sure what exactly is up with my setup here.

Anyway thanks for the clarification!