moonlight-stream/moonlight-embedded #566: Build error imx.c on Ubuntu 16.04

MrCapone, 2017-09-22:
**Moonlight Embedded version:** 2.3.4 + c8a8f05
**Moonlight Embedded source:** https://github.com/irtimmer/moonlight-embedded
**Moonlight Embedded build running on:** Udoo quad
**Moonlight Embedded running on distribution:** Ubuntu 16.04

**output**
```
[ 95%] Built target moonlight
In file included from /home/ubuntu/dev/moonlight-embedded/src/video/imx.c:38:0:
/usr/include/linux/mxc_v4l2.h:57:2: error: unknown type name ‘__u32’
  __u32   type; /* enum v4l2_buf_type */
  ^
CMakeFiles/moonlight-imx.dir/build.make:62: recipe for target 'CMakeFiles/moonlight-imx.dir/src/video/imx.c.o' failed
make[2]: *** [CMakeFiles/moonlight-imx.dir/src/video/imx.c.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/moonlight-imx.dir/all' failed
make[1]: *** [CMakeFiles/moonlight-imx.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

**Possible solution:** include `<linux/types.h>` before `<linux/mxc_v4l2.h>` on `moonlight-embedded/src/video/imx.c`

irtimmer, 2017-09-22:
What is the content of /usr/include/linux/mxc_v4l2.h? As no single version of this file that I have found contains the line `__u32   type; /* enum v4l2_buf_type */`.

MrCapone, 2017-09-24:
My board is Udoo Quad. I use [custom linux kernel](https://github.com/fmntf/linux_kernel/) for this platform. Ubuntu 16.04 minfs for armhf. All imx6 specific libs and header i installed from [udoo team repository](http://repository-test.udoo.org/xenial/). _/usr/include/linux/mxc_v4l2.h_ provides by _linux-libc-dev_ package from this repo. I alredy successfully built moonlight by including `<linux/types.h>`, that also provides by _linux-libc-dev_ package. You can check commit from my fork https://github.com/MrCapone/moonlight-embedded/commit/9ed221bda8cb194a554e21785be5fbfbd6477848

MrCapone, 2017-09-24:
Here file content https://gist.github.com/MrCapone/c314ea27ff50d129adf5afc0d9a0a97e

irtimmer, 2017-10-10:
Pull request #564 have been merged

closed 2017-10-10 by irtimmer