moonlight-stream/moonlight-embedded #568: Pi3 installation from jessie source -> Depends: libssl1.0.0 (>= 1.0.0) but it is not installable

victorator, 2017-09-28:
The following happens on Raspbian with Raspberry Pi 3 fresh install with the following line added to sources.list: deb http://archive.itimmer.nl/raspbian/moonlight jessie main
and apt-get update

**pi@raspberrypi:~ $ sudo apt-get install moonlight-embedded
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 moonlight-embedded : Depends: libssl1.0.0 (>= 1.0.0) but it is not installable
E: Unable to correct problems, you have held broken packages.**

benwilesmusic, 2017-09-29:
I've spent the last few days figuring this out, and the best way around this problem seems to be to install over raspbian jessie, rather than the latest raspbian stretch. Libssl1.0.0 is not available in the raspbian stretch repository, and moonlight-embedded seems to demand that version.

The last published version of raspbian jessie is here as an image file: http://downloads.raspberrypi.org/raspbian/images/raspbian-2017-07-05/

Another possible solution is to try adding another repository to the end of sources.list; the best bet might be the old raspbian jessie repository – I think it worked for me: 

deb http://archive.raspbian.org/raspbian jessie main contrib non-free

You should probably delete this entry or comment it out with a # at the start of the line after you have installed moonlight-embedded, as you don't want apt to use it for anything else.

Another possibility would be to add debian backports (which will only work for, I think, the pi 2 and pi 3, neither of which I have):

deb http://ftp.debian.org/debian stretch-backports main

Alternatively you could compile libssl1.0.0 yourself, but that seems like a lot of bother to me.

I ended up installing on top of retropie, which is currently still based on jessie, and it works fine.

irtimmer, 2017-09-29:
Duplicate of #549

closed 2017-09-29 by irtimmer

victorator, 2017-09-29:
Many thanks. I tried installing it manually, then from the sources you mentioned. Each time I get a pairing error :(

Connect to 192.168.55.99...
Please enter the following PIN on the target PC: 7433
Failed to pair to server: Failed to sign data

victorator, 2017-10-02:
solved with sudo !

GeoZzNL, 2017-10-24:
Solved with adding the following line in /etc/apt/sources.list/:

deb http://archive.raspbian.org/raspbian jessie main contrib non-free
AND
deb http://archive.itimmer.nl/raspbian/moonlight jessie main

Then did;
1. Sudo -i
2. Apt update
3. Apt install moonlight embedded
4. Apt update 
5. Apt upgrade
6. Restart raspberry
7. Sudo Moonlight pair 192.xxx.xxx.xxx
8. Done.

sharatkanthi, 2018-04-16:
This is because the web socket library is not installed. Try running 

`sudo apt-get install libwebsockets-dev`
before running 

`sudo apt-get install mosquitto`