moonlight-stream/moonlight-embedded #570: i965_drv_video.so init failed

mzernetsch, 2017-10-01:
**_NVidia Geforce Experience version:_** 3.9.0.61
**Moonlight Embedded version:** 2.4.1
**Moonlight Embedded source:** https://aur.archlinux.org/packages/moonlight-embedded/
**Moonlight Embedded running on:** Ultrabook (Intel i5 6200U)
**Moonlight Embedded running on distribution:** Arch Linux

**Verbose output `-verbose` of Moonlight Embedded:**

Output without LIBVA_DRIVER_NAME:
```bash
[user@arch ~]$ moonlight stream -verbose 10.0.0.4
Moonlight Embedded 2.4.1 (SDL;X11;VDPAU;VAAPI;EMBEDDED;ALSA;PULSE)
Connect to 10.0.0.4...
NVIDIA GeForce GTX 770, GFE 3.9.0.61 (gs_04_00_22663502, 7.1.400.0)
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[AVHWDeviceContext @ 0x55953e847dc0] Failed to initialise VAAPI connection: -1 (unknown libva error).
Failed to open VDPAU backend libvdpau_nvidia.so: cannot open shared object file: No such file or directory
Platform X Window System (software decoding)
Loading mappingfile /usr/share///moonlight/gamecontrollerdb.txt
Failed to open device /dev/input/event3
Failed to open device /dev/input/event4
Failed to open device /dev/input/event9
Failed to open device /dev/input/event1
Failed to open device /dev/input/event2
Failed to open device /dev/input/event6
Failed to open device /dev/input/event8
Failed to open device /dev/input/event10
Failed to open device /dev/input/event11
Failed to open device /dev/input/event12
Failed to open device /dev/input/event13
Failed to open device /dev/input/event14
Failed to open device /dev/input/event15
Failed to open device /dev/input/event16
Failed to open device /dev/input/event7
Failed to open device /dev/input/event0
Failed to open device /dev/input/event5
Failed to open device /dev/input/event17
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 27146, but got 27147)
Decode failed - Invalid data found when processing input
Decode failed - Invalid data found when processing input
Unrecoverable frame 3: 16+0=16 received < 24 needed
Network dropped an entire frame
Returning RTP packet queued for too long
Received OOS audio data (expected 27179, but got 27181)
Invalidate reference frame request sent (3 to 3)
```

Output with LIBVA_DRIVER_NAME=i965:
```bash
[user@arch ~]$ export LIBVA_DRIVER_NAME=i965
[user@arch ~]$ moonlight stream -verbose 10.0.0.4
Moonlight Embedded 2.4.1 (SDL;X11;VDPAU;VAAPI;EMBEDDED;ALSA;PULSE)
Connect to 10.0.0.4...
NVIDIA GeForce GTX 770, GFE 3.9.0.61 (gs_04_00_22663502, 7.1.400.0)
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns -1
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva error: /usr/lib/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
[AVHWDeviceContext @ 0x561c80bf6b40] Failed to initialise VAAPI connection: -1 (unknown libva error).
Failed to open VDPAU backend libvdpau_nvidia.so: cannot open shared object file: No such file or directory
Platform X Window System (software decoding)
Loading mappingfile /usr/share///moonlight/gamecontrollerdb.txt
Failed to open device /dev/input/event3
Failed to open device /dev/input/event4
Failed to open device /dev/input/event9
Failed to open device /dev/input/event1
Failed to open device /dev/input/event2
Failed to open device /dev/input/event6
Failed to open device /dev/input/event8
Failed to open device /dev/input/event10
Failed to open device /dev/input/event11
Failed to open device /dev/input/event12
Failed to open device /dev/input/event13
Failed to open device /dev/input/event14
Failed to open device /dev/input/event15
Failed to open device /dev/input/event16
Failed to open device /dev/input/event7
Failed to open device /dev/input/event0
Failed to open device /dev/input/event5
Failed to open device /dev/input/event17
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 40434, but got 40435)
Decode failed - Invalid data found when processing input
Decode failed - Invalid data found when processing input
Unrecoverable frame 3: 22+0=22 received < 24 needed
Network dropped an entire frame
Invalidate reference frame request sent (3 to 3)
Returning RTP packet queued for too long
Received OOS audio data (expected 40463, but got 40465)
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done
```

**What is the expected result?**
moonlight should start streaming using hardware decoding (vaapi).

**What happens instead of that?**
moonlight starts streaming without hardware decoding.
initialization of vaapi fails.

**Additional information:**
When i test using vainfo it seems to work properly:
```bash
[user@arch ~]$ vainfo 
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva info: va_openDriver() returns 0
vainfo: VA-API version: 0.40 (libva )
vainfo: Driver version: Intel i965 driver for Intel(R) Skylake - 1.8.3
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
```
I am using Wayland. Could that be a problem?

mzernetsch, 2017-10-15:
I tested it in a x11 session this weekend and it worked properly. So basically hardware decoding (using vaapi) doesn't work in a wayland session.

FockeOeuf, 2017-10-25:
Currently addons in Open/LibreElec do not give access to the platform option in the xml configuration file.
So on an Intel NUC (Haswell), what platform will be used?

How is the platform choice performed during launch without any value specified for the platform. Can forcing the X11 platform in the command line force the VAAPI hardware decoding?