moonlight-stream/moonlight-embedded #572: Mapping for Thrustmaster Thrustmaster dual analog 3.2 (030000004f04000015b3000010010000) on /dev/input/event2 is incorrect

MateEke, 2017-10-06:
Please provide the following info.

**_NVidia Geforce Experience version: GFE 3.9.0.97
**Moonlight Embedded version: 2.4.3
**Moonlight Embedded source: repository
**Moonlight Embedded running on: Raspberry Pi 3
**Moonlight Embedded running on distribution: Raspbian Jessie/Stretch

**Verbose output `-verbose` of Moonlight Embedded:**
```
Moonlight Embedded 2.4.3 (EMBEDDED;ALSA;PI)
Connect to 192.168.2.100...
NVIDIA GeForce GTX 1080, GFE 3.9.0.97 (gs_04_00_22663502, 7.1.400.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Detected Thrustmaster Thrustmaster dual analog 3.2 (030000004f04000015b3000010010000) on /dev/input/event2 as Thrustmaster Dual Analog 4
Mapping for Thrustmaster Thrustmaster dual analog 3.2 (030000004f04000015b3000010010000) on /dev/input/event2 is incorrect
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 315, but got 316)
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done
```

**What is the expected result?**
Correct controller handling

**What happens instead of that?**
I have got a Thrustmaster dual analog 4 controller, and when I start Moonlight there is this error: "Mapping for Thrustmaster Thrustmaster dual analog 3.2 (030000004f04000015b3000010010000) on /dev/input/event2 is incorrect". As I see the mapping is correct (tried jstest) but left and right trigger doesn't work in Steam and games (every other buttons works fine).  I have tried clean install of jessie and stretch, but the issue persists on both of them.

MateEke, 2017-10-06:
I have tried an old Canyon controller, it is detected as "USB Gamepad", but there is the same issue - Mapping for USB Gamepad (03000000100800000300000010010000) on /dev/input/event2 is incorrect

Static1589, 2017-10-09:
Edit on above post.

Sorry, R2 / Right Trigger is recognized as Start.

Kazatel, 2017-12-14:
Check my post, maybe you will be able to correct mapping. 
https://github.com/irtimmer/moonlight-embedded/issues/603

MateEke, 2017-12-14:
I have bought a Steam Link, and the Rpi is running Volumio, so I'm not using moonlight anymore.