moonlight-stream/moonlight-embedded #573: Vero 4K support

TheHacker66, 2017-10-07:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.9.0.97
**Moonlight Embedded version:** 2.4.3
**Moonlight Embedded source:** _compiled from source_
**Moonlight Embedded running on:** _Vero 4K_
**Moonlight Embedded running on distribution:** _OSMC_

**Verbose output `-verbose` of Moonlight Embedded:**

> Searching for server...
> Connect to 192.168.178.20...
> No mapping available for meson-ir (19000000000000000000000000000000) on /dev/input/event2
> No mapping available for gpio_keypad (10000000010000000100000000010000) on /dev/input/event0
> No mapping available for cec_input (100000008e1b0000ec0c000001000000) on /dev/input/event1
> Initializing platform...done
> Resolving host name...done
> Starting RTSP handshake...done
> Initializing control stream...done
> Initializing video stream...done
> Initializing audio stream...done
> Initializing input stream...done
> Starting control stream...done
> OPEN es DEVICE
> codec_init amstream version : 2.0
> [codec_init 732]type =1
> [codec_init 740]
> Starting video stream...done
> ALSA lib pcm_direct.c:896:(snd1_pcm_direct_initialize_slave) slave plugin does not support mmap interleaved or mmap noninterleaved access
> ALSA lib pcm_dmix.c:1034:(snd_pcm_dmix_open) unable to initialize slave
> Starting audio stream...Alsa error code -22
> Audio stream start failed: -1
> Stopping video stream...ENet wait interrupted
> Control stream connection failed
> Loss Stats: Transaction failed: 11
> [codec_close 851]
> [codec_close_cntl]video codec close return=0!
> done
> Stopping control stream...done
> Cleaning up input stream...done
> Cleaning up audio stream...done
> Cleaning up video stream...done
> Cleaning up control stream...done
> Cleaning up platform...done
> EVIOCGRAB failed with error 16
> EVIOCGRAB failed with error 16

TheHacker66, 2017-11-06:
I'm bumping this and adding more logs:

https://gist.github.com/antsu/b426d72a35e7b32fbad64f05552225ba

Hope that someone will look into this sooner or later ;)

cgutman, 2017-11-20:
Can you try specifying the target audio device manually?

If that fails, try building with PulseAudio to see if the PA backend works.

TheHacker66, 2017-11-20:
Hi @cgutman!

Thanks for the suggestions, specifying the audio works (kinda), i can hear crackling sounds coming from the tv but the video is still black.

Something interesting coming from the verbose output:

`[codec_init 732]type =1
[codec_init 740]
Starting video stream...done
Starting audio stream...done
Starting input stream...done
EVIOCGRAB failed with error 16
EVIOCGRAB failed with error 16
Alsa error from writei: -32
Returning RTP packet queued for too long
Received OOS audio data (expected 28132, but got 28133)Waiting for IDR frame
Alsa shortm write, write 48 frames
Alsa shortm write, write 64 frames
Alsa shortm write, write 192 frames
Alsa shortm write, write 64 frames
Alsa error from writei: -32
Unrecoverable frame 197: 16+2=18 received < 21 needed
Network dropped an entire frame
codec_write error: ffffffff
[codec_close 851]
[codec_close_cntl]video codec close return=0!
OPEN es DEVICE
[codec_init 732]type =1
[codec_init 740]
[codec_reset:1199]ret=0
Requesting IDR frame on behalf of DR
Waiting for IDR frame
Waiting for IDR frame
Alsa error from writei: -32
...`

Hope this might give some clues!

cgutman, 2017-12-06:
What exact parameters are you passing to stream? What GPU is in your PC?

You might try adding `-codec h264`. I found that some Amlogic chips don't like the default h.265 stream that GeForce Experience produces.

I implemented a workaround in moonlight-common-c for the most recent GFE version, which supports specifying the number of reference frames. https://github.com/moonlight-stream/moonlight-common-c/commit/f6ae7fc2b4d2c0c5d2a8cb8ebc55f4157156c9f1

TheHacker66, 2017-12-06:
The last time I simply ran `moonlight stream -audio hw:0,0 -platform aml`

I merged the recent commits, I'll compile, test using your suggestions and report back tonight. Thanks!

TheHacker66, 2017-12-07:
Unfortunately specifying the h264 codec didn't change much, i got less errors in console but the screen is still black.

TheHacker66, 2018-01-17:
OSMC and Vero 4k were updated to Stretch. Unfortunately the black screen issue still remains.

samnazarko, 2018-01-17:
Can you try run strace and see if it gives us anything useful?

TheHacker66, 2018-01-17:
Yup, here's the log: https://paste.osmc.tv/isavozohoz

danielfmo, 2018-01-17:
Hello,

I belibe this is a kodi and amlogic problem and not with moonlight.

Please take a look here:
https://forum.odroid.com/viewtopic.php?f=91&t=15456&p=171714&hilit=%2Fsys%2Fclass%2Fvideo%2Fdisable_video#p171714

try to set /sys/class/video/disable_video to 0 before starting moonlight

samnazarko, 2018-01-17:
Good find @danielfmo . Let us know how it goes @TheHacker66 .
I can patch Kodi to reset this on exit; or you can set it in a startup script in Moonlight.

TheHacker66, 2018-01-18:
Well guys, unfortunately setting that flag before running moonlight didn't change anything. I can run another strace with the flag set to 0 if needed.

So close yet so far..

samnazarko, 2018-02-14:
@TheHacker66 I've sent you a PM with a kernel with amlvideo debugging on. It may give us some clues if the decoder isn't happy.

TheHacker66, 2018-02-14:
I won't be at home till Saturday, so I'll test it once I get back.

samnazarko, 2018-02-14:
Cool. I found why HEVC wasn't tolerated by Moonlight (buf margin needs a bump). This also fixed a couple of playback issues.

Sent from a mobile device, so please excuse the brevity of this reply

TheHacker66, 2018-02-21:
Hi, sorry for not replying earlier.

I installed the kernel and rebooted. Unfortunately there seems to be an incompatibility with the latest GFE version because moonlight complains about an udp port which is open in the firewall and that was working before. The version I have installed is 3.12.0.84

I tried uninstalling that and installing the version available on nvidia's site (3.12.0.79) to no avail. Deleting firewall rules (that were recreated after the install) didn't change anything.

I'll try to reinstall a even older version to see if the problem goes away.

TheHacker66, 2018-02-21:
Ok, I reinstalled the version I had when I started this thread (3.9.0.97) and moonlight started working again.

Here's the dmesg log with disable_video set to 1:
https://paste.osmc.tv/ezeludacig

This is the dmesg log with the flag set to 0:
https://paste.osmc.tv/milavatana

samnazarko, 2018-02-21:
Hopefully you're feeling better now.

Thanks for giving this a go. 

>[  +0.010327] codec:vdec1 video changed to 0 x 0 0 fps clk->200MHZ

I think this is the problem. The video decoder doesn't seem to be set up correctly. 

I believe Moonlight uses libamcodec. Can you confirm if you compiled Moonlight using the libamcodec-dev-osmc package?

TheHacker66, 2018-02-22:
Hi Sam, thanks.

I have indeed installed vero3-libamcodec-dev-osmc. I modified the `cmake/FindAmlogic.cmake` file to point to `/usr/osmc/lib/` where libamcodec.so is located.

It is correct?

TheHacker66, 2018-02-22:
Here's the output from moonlight:

http://paste.osmc.tv/poqayipobo.vbs

samnazarko, 2018-02-22:
Yes -- that's one way to get the compile to use the OSMC versions of libamcodec.

>[  +0.007596] codec:vdec1 video changed to 0 x 0 0 fps clk->200MHZ

This still looks to be the problem. I'm not sure why the decoder is not getting hints properly; but I suspect something in Moonlight isn't setting up the amprivate structs properly.

samnazarko, 2018-02-22:
@cgutman @danielfmo might have some more ideas.

I'll see if I can get access to an NVIDIA system to test this out with.

TheHacker66, 2018-02-22:
You should still have remote access to my Vero 4k to fiddle with. I won't be at home tomorrow but I will be this weekend if you need me for tests.

williamluke4, 2018-03-12:
@TheHacker66 Have you had any luck?

TheHacker66, 2018-03-20:
Nope. I think we need @cgutman or @irtimmer help.

samnazarko, 2018-03-24:
We're happy to get hardware out to resolve this.

samnazarko, 2018-06-24:
Nothing yet unfortunately.
I can't think of anything we have significantly diverged with that would cause such issues. 

I'm happy to ship hardware to any developers. In the interim, I'm going to see if I can set up an NVIDIA system to look in to this myself.  I am away for a few weeks however.

ZacWolf, 2018-07-23:
Anything on this yet?

samnazarko, 2018-07-23:
Unfortunately nothing has moved this forward yet.

maxanier, 2018-09-30:
Finally got around testing it myself on my new Vero4k+, but I got the same issue: Black screen, but some crackling audio.

(@samnazarko ) I am using the OSMC August update with the vero4k+ video improvements installed.

I have compiled moonlight-embedded from the master branch, but adjusted cmake/FindAmlogic.cmake to include the /opt/vero3/include and /usr/osmc/lib paths ([gist](https://gist.github.com/maxanier/3318bb1d00c73a935933f2272bd53df7)).

I tried 720,1080,4k as well as h264 and h265. Result is the same.
Logs look similar to TheHacker66 ones:
moonlight stream -platform aml -localaudio -codec h265 -1080 -verbose: [MoonlightLog](https://gist.github.com/maxanier/5849be0e6c3c71c649599b29b684ecb9) [dmesg](https://gist.github.com/maxanier/67169985310c44096aac27e8f019553d)

I tried to do some debugging, but the only thing I could figure out so far is that the `codec_write error: ffffffff` log message is caused by the codec  handle is temporarily unavailable (11 EAGAIN) in amcodec `codec_write` but this is probably normal behaviour. 
I do not really know where to continue looking since I can't really tell if the problem lies in moonlight (embedded/common-c) or in amcodec or somewhere else in the vero4k binaries.

samnazarko, 2018-09-30:
Glad to hear you've received the new device.

Did you stop Kodi first? sudo systemctl stop mediacenter

maxanier, 2018-09-30:
Yes

samnazarko, 2018-11-20:
I understand. We need some assistance from the Moonlight guys here. Hopefully there is an opportunity to do this.

cgutman, 2018-11-20:
The IDR frame warnings are a symptom of the problem. When `codec_write()` fails, the error triggers the IDR frame request to resync with the host (since video data was lost).

Can you see whether it's failing on the first `codec_write()` call or a later one? The AML decoder may expect to receive an entire frame at a time, rather than split up as we are currently submitting them. Can you try modifying the code to submit the entire frame at once like the SDL backend does? https://github.com/irtimmer/moonlight-embedded/blob/master/src/video/sdl.c#L66

From this Kodi fork, it looks like they are explicitly checking for EAGAIN and continuing to retry until it clears. You could also try that with or without the above change too. https://gitlab.com/ember-dev/kodi/blob/46dc687ff8c2ea711021b4b98a562eb9ce21a4d8/xbmc/cores/dvdplayer/DVDCodecs/Video/AMLCodec.cpp#L679

maxanier, 2018-11-21:
Thank you very much for your response.

I implemented your first suggestion and it seems to fix the codec_write errors. Moonlight stops complaining.
However, the screen stays black. I still get the crackling (maybe slowed down) steam start sound (unless I use -localaudio).

[Updated aml.c](http://termbin.com/4b8z)
[Moonlight Verbose Output ](https://gist.github.com/maxanier/ba9511e79ba8611280d2b75e5ea18d94#file-moonlightlog)
[Dmesg](https://gist.github.com/maxanier/ba9511e79ba8611280d2b75e5ea18d94#file-dmesg)

I cannot tell much from it, but one thing that confuses me is a "init_decode_head_hw: (1920, 1088)", but I do not think this is related to our problem.

Just out of curiosity I logged the size of the frames passed to codec_write ([output](https://gist.githubusercontent.com/maxanier/ba9511e79ba8611280d2b75e5ea18d94/raw/e83ef0b0eaa35cef049a2f85471cb9d4046cd19b/write%2520log)). The output looks reasonable to me, about 30 logs per second, and the size starts large and gets smaller, probably because the transmitted frames is not changing (Steam start screen). But that are just guesses.

I also tried playing around with /sys/class/video/disable_video and more, but it does not help.

@samnazarko maybe something in osmc is preventing the decoded stream from actually being displayed. 
After stopping kodi, it displays the console output (from my rc.local), but after starting/stopping moonlight everything stays black until I restart kodi.

samnazarko, 2018-11-22:
See https://github.com/osmc/libamcodec/blob/2017.10/amplayer/player/player_av.c#L1572 for how to set up decoder.

>I also tried playing around with /sys/class/video/disable_video and more, but it does not help.

You could try booting with Kodi disabled (systemctl disable mediacenter); in case Kodi or our watchdog are causing issues.

maxanier, 2018-11-23:
Unfortunately, disabling kodi did not help.

I had a look at the code you linked, but could see anything interesting. I do not have any experience with video en/decoding, so I might have missed something relevant.

After that I had a look at https://github.com/osmc/libamcodec/blob/master/examples/esplayer.c which setups framebuffer and axis, but that did not help either.

Is there any way to get more debug log from the libs or kernel parts responsible for actually displaying what is passed to `codec_write`?

samnazarko, 2018-12-12:
Sorry for the late reply on this.

I was able to modify the examples Makefile target and build simple_player.
With Big Buck Bunny I can initialise the VPU and see frames being decoded; but I only get a black screen. 

It's probably worth comparing the framebuffer state (sys/fb) when playing a video via Kodi vs trying to use Moonlight.

maxanier, 2019-02-21:
Finally got around to try it again.
I updated the Vero to the December update, pulled all the latest upstream changes of moonlight-embedded and rebuild. But the issue remains.

I tried comparing the state of `/sys/class/graphics/fb0,1 /state,mode, ...` as well as `/sys/class/display/axis,mode`. But since I couldn't find any helpful documentation on the dozens of things I am not smarter than before. 
The only thing I noticed is that `/sys/class/graphics/axis` and `/sys/class/graphics/mode` are not updated if I stop kodi and start moonlight. The values of the last video playback remain, even if moonlight runs on a different resolution.

I am also not sure if it has any effect to directly modify those (e.g. `echo 2 > /sys/class/display/disable_video`). The values are update, but I never manged to see any difference on screen.

TheHacker66, 2019-02-24:
Hey everyone, I also updated my fork and tried to no avail. Saw that some aml libraries were added to the cmake and fixed the paths for osmc but it didn't make a change.

hope @irtimmer or @cgutman have some other clue.

maxanier, 2019-04-09:
I decided to give up on this. It feels like we are close, but without some experience with the Linux display system it is rather difficult to get it to work.
I am back to RPi3 for anything non 4k related, because neither moonlight nor hyperion work (well) on the vero.

samnazarko, 2019-04-09:
Hyperion should work well in the next update.
Unfortunately I'm not sure how to proceed with Moonlight.

EvgeniySpinov, 2021-05-05:
I'd like to join the battle for the moonlight + vero, cause I do not want to deal with 1080 on raspberry pi.

@samnazarko Is there any way to emulate Vero 4K+? I have only 1 device and it's used in living room and family would not be happy if I would take it for experiments. 

@maxanier Were you able to do super-verbose output by moonlight on connect (besides strace)? Is there such a mode? On your raspberry, you're also running OSMC? Upstream version and everything works? I also have Raspberry pi 3 where moonlight works without any hick ups. I have older version of the OSMC there though (Kodi 17.x).

samnazarko, 2021-05-05:
>Is there any way to emulate Vero 4K+?

Not really unfortunately.

cgutman, 2021-05-05:
Does the Vero 4K support the standard V4L2 stateful/stateless decoding APIs? If so, that's probably way to go, but Moonlight Embedded doesn't have a standard V4L2 backend yet. V4L2 is the way forward to get out of the proprietary vendor-specific API hell that is currently the embedded world.

The V4L2 stateful API is supported in upstream FFmpeg (`h264_v4l2m2m` and `hevc_v4l2m2m` decoders), but the V4L2 stateless API requires [out-of-tree patches](https://github.com/LibreELEC/LibreELEC.tv/tree/master/packages/multimedia/ffmpeg/patches/v4l2-request) (`v4l2request` hwaccel) while the kernel interface is still being completed. Which one you will need depends on how the Amlogic V4L2 kernel drivers are written.

If you want to test V4L2 on the Vero 4K, you can try building [moonlight-qt](https://github.com/moonlight-stream/moonlight-qt) with a V4L2-enabled FFmpeg build. Then you can test V4L2+EGL or V4L2+DRM rendering and see if performance is acceptable. Users on the RK3288/RK3399 with V4L2+EGL and V4L2+DRM on the Pi 4 for HEVC have reported good performance.

If V4L2 does work well, it shouldn't be too hard to drop in similar code in Moonlight Embedded to what Moonlight Qt. Moonlight Embedded doesn't have overlay support or windowed support (on most devices), so it's probably sufficient to just implement the V4L2+DRM path and omit V4L2+EGL.

renatobellotti, 2022-08-12:
Has there been progress? :)

samnazarko, 2022-08-12:
Unfortunately not.

cgutman, 2022-08-13:
If the hardware offer still stands, I'd be happy to take a look.

Is there an FFmpeg build that supports hardware decoding on the Vero's current current media stack? FFmpeg would probably be easiest to integrate because Moonlight Qt supports it already.

Do you know what APIs the Kodi folks are using on this hardware? Is it V4L2+DRM or the old amcodec stuff?

samnazarko, 2022-08-13:
Hi - we still use AMCodec on this device. It’s possible to support AMCodec via ffmpeg with some hacks but this hasn’t been tested. 

AML’s current focus is a V4L2 GStreamer stack. 

Happy to send some hardware.

cgutman, 2022-08-14:
> Hi - we still use AMCodec on this device. It’s possible to support AMCodec via ffmpeg with some hacks but this hasn’t been tested.

So it sounds like our existing [amcodec support](https://github.com/moonlight-stream/moonlight-embedded/blob/master/src/video/aml.c) should be expected to work then. Hopefully it is something relatively simple.

> Happy to send some hardware.

OK, please reach out to cameron@moonlight-stream.org and we can get the shipping details sorted out.

TheHacker66, 2022-08-15:
@cgutman I tried syncing and compiling my fork yesterday to see if updating to Debian 11 (which will be released soon on OSMC) changed anything but the same issue occurs.

Here's the fork https://github.com/TheHacker66/moonlight-embedded

Let me know if there's something I can do while you wait to receive the hardware.

Thanks for not abandoning OSMC support!

samnazarko, 2022-08-15:
Thanks, I’ll get in touch with you shortly.

samnazarko, 2022-09-02:
Hi -- I was travelling so it took a bit longer than expected to email you, but I've done so now and put a unit aside for you. Please let me know if you've received my email

cgutman, 2022-09-03:
Yep, I got it and just replied. It got in my spam filter, so I didn't see it earlier.

samnazarko, 2022-09-07:
This has shipped and I sent you tracking details. There’s currently a postal strike for two days in the UK, so I wouldn’t expect the package to move much until next week

cgutman, 2022-10-13:
I worked on this for a few hours today trying various things and couldn't get anything working.

To see if it was Moonlight-specific, I tried the samples from the [OSMC Amcodec](https://github.com/osmc/libamcodec) repo and those exhibited the same symptoms (reporting successful decoding but not displaying anything).

The log messages from the kernel seem to be as expected:
```
[   78.416045] vdec_init, dev_name:ammvdec_h265, vdec_type=VDEC_TYPE_STREAM_PARSER  id = 0
[   78.416050] vdec_init set vfm decoder 00000000428cad09
[   78.416862] HEVC stbuf alloced at 00000000f2624b3e, secure = 0, size = 10485760
[   78.416977] pts_start, type=2
[   78.417534] video first pts = 0
[   78.417542] vdec_request_irq 00000000a9a08f5a, parser
[   78.418868] decoder_bmmu_box_alloc_box, tvp_flags = 0
[   78.419082] [0]dynamic_buf_num_margin=7
[   78.419088] [0]double_write_mode=0
[   78.419097] [0]h265:pts_unstable=0
[   78.419102] [0]h265: ver (201602101,0) decinfo: 1280x720 rate=1600
[   78.420499] [TEE], the fw (hevc_mmu_swap) will be loaded.
[   78.420602] hevc mux clock is 666666666 Hz
[   78.420608] hevc video changed to 3840 x 2160 60 fps clk->667MHZ
[   78.420898] ppmgr local_init
[   78.420911] di_receiver_event_fun: vframe provider reg ppmgr
[   78.421990] set run_early_proc_fun_flag to 1
[   78.422000] vpu_clkb = 0.
[   78.422001] DI: reg f
[   78.425832] ppmgr local_init
[   78.426125] vdec_init, vf_provider_name = decoder, b 0
[   78.426153] vdec->port_flag=0x102, port_flag=0x10b
[   78.451041] osmc videoenhancement: session context configured successfully
[   78.478751] [0]vh265 cached=16384  need_size=16384 speed= 60 ms
[   78.479573] [0]hevc->performance_profile 0
[   78.479587] [0]set pic_list_init_flag 1
[   78.481705] [0]init_buf_spec2 1280 736
[   78.481712] [0]set pic_list_init_flag to 2
[   78.481739] [0]set pic_list_init_flag to 3
[   78.481742] [vdec_kpi][vh265_isr_thread_fn] First I frame coming.
[   78.484248] [0]VH265: output first frame
[   78.484269] [vdec_kpi][post_video_frame] First I frame decoded.
[   78.484337] DI bypass all 1280x720-0x501000.
[   78.511486] cal_dur_from_pts inner driver dur1600 
[   78.533530] vframe_rate_uevent: sent uevent FRAME_RATE_HINT=1600
```

The `[   78.484248] [0]VH265: output first frame` line in particular seems to indicate that it did successfully output a frame to _somewhere_.

I also added a call to `codec_get_vdec_state()` before submitting each frame to confirm that the decoder is not reporting errors and it doesn't seem to be.

```
Received first video packet after 0 ms
Stream info: 0x0x60 | Errors: 0 | Status: 255
Stream info: 0x0x60 | Errors: 0 | Status: 255
Stream info: 1280x736x60 | Errors: 0 | Status: 255
```

The first two have no dimensions because the decoder hasn't processed enough of the bitstream yet, but then it does initialize (further indicate that the codec is indeed parsing the actual bitstream correctly and it's an output-related issue).

@samnazarko Is there something special that needs to be done to get amcodec output working? Since even the amcodec samples fail to play, I think this is probably related to Vero-specific kernel or amcodec changes.

I also tried moonlight-qt to see if maybe it could work more easily, but it looks like the Vero uses a legacy fbdev driver which is not something moonlight-qt supports. Even if it did support fbdev, it would not be able to do the performant zero-copy rendering that we can do with EGL or DRM renderers, so the performance likely wouldn't be good.

samnazarko, 2022-10-15:
Hi,

The amcodec in that repository is quite old and we've a much more improved one which is installed in OSMC automatically. You can get the appropriate headers with ```sudo apt-get install libamcodec-dev-osmc```. That's just a newer library itself -- the examples are the same and unchanged.

samnazarko, 2022-10-15:
According to @tanio99

>It might be related to the decoder type being used when initializing libamcodec. For HEVC playback one needs to set

codecParam.decoder_type = DECODER_TYPE_FRAME_MODE;

cgutman, 2022-10-31:
> The amcodec in that repository is quite old and we've a much more improved one which is installed in OSMC automatically. You can get the appropriate headers with sudo apt-get install libamcodec-dev-osmc. That's just a newer library itself -- the examples are the same and unchanged.

Is the source for that available somewhere? I found it useful to look at the old one to try to figure out what the various functions and fields do.

> I assume you stopped Kodi first: sudo systemctl stop mediacenter

Yep

> It might be related to the decoder type being used when initializing libamcodec. For HEVC playback one needs to set `codecParam.decoder_type = DECODER_TYPE_FRAME_MODE`

When I set DECODER_TYPE_FRAME_MODE for HEVC, it seems to break even worse. `dmesg` fills up with `[0]cur lcu idx = 0, (total 920), set error_mark` messages and the error count as reported by `codec_get_vdec_state()` increases with each frame.

```
[  386.289232] the demux           clock on, ref cnt: 1
[  386.289469] the parser_top      clock on, ref cnt: 1
[  386.289474] the vdec            clock on, ref cnt: 1
[  386.289498] the clk_hevc_mux    clock on, ref cnt: 1
[  386.289508] hevc mux clock is 500000000 Hz
[  386.289623] vdec_create instance 000000004e80c985, total 1, PM: legacy
[  386.289722] no drmmode
[  386.289786] vdec_init, dev_name:ammvdec_h265, vdec_type=VDEC_TYPE_FRAME_BLOCK  id = 0
[  386.292733] decoder_bmmu_box_alloc_box, tvp_flags = 0
[  386.293169] [0]dynamic_buf_num_margin=7
[  386.293175] [0]double_write_mode=0
[  386.293183] [0]h265:pts_unstable=0
[  386.293189] [0]h265: ver (201602101,0) decinfo: 1280x720 rate=1600
[  386.293940] [TEE], the fw (hevc_mmu_swap) will be loaded.
[  386.294037] hevc mux clock is 666666666 Hz
[  386.294041] hevc video changed to 3840 x 2160 60 fps clk->667MHZ
[  386.294287] vfm map  created
[  386.294334] vf_reg_provider Error to notify receiver
[  386.294338] Error: vf_notify_receiver, fail to get receiver of provider vdec.h265.00
[  386.294341] Error: vf_notify_receiver, fail to get receiver of provider vdec.h265.00
[  386.294345] vdec_init, vf_provider_name = vdec.h265.00, b 0
[  386.320143] vdec->port_flag=0x102, port_flag=0x10b
[  386.324283] osmc videoenhancement: session context configured successfully
[  386.338968] [0]vh265 cached=16384  need_size=16384 speed= 40 ms
[  386.341230] [0]hevc->performance_profile 0
[  386.341253] [0]set pic_list_init_flag 1
[  386.343744] [0]init_buf_spec2 1280 736
[  386.343752] [0]set pic_list_init_flag to 2
[  386.343793] [0]set pic_list_init_flag to 3
[  386.343796] [vdec_kpi][vh265_isr_thread_fn] First I frame coming.
[  386.549268] [0]timeout_process decoder timeout
[  386.549277] [0]cur lcu idx = 151, (total 920), set error_mark
[  386.549318] [0]cur lcu idx = 151, (total 920), set error_mark
[  386.549813] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.549817] [0]VH265: output first frame
[  386.549837] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.550312] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.550754] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.550758] [0]VH265: output first frame
[  386.550808] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.551277] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.551283] [0]VH265: output first frame
[  386.551306] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.551753] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.551756] [0]VH265: output first frame
[  386.551773] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.552244] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.552665] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.552669] [0]VH265: output first frame
[  386.552684] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.553146] [0]cur lcu idx = 0, (total 920), set error_mark
[  386.553567] [0]cur lcu idx = 0, (total 920), set error_mark
```

> Additionally, the VFM map doesn't seem to be set in the logs.

Hmm, what's a VFM map?

> You should initialise the codec parameter structure fully, i.e:

This is what I'm doing now but no luck so far: https://github.com/moonlight-stream/moonlight-embedded/commit/87cad37958210dbcda2d2f14cccc99ad72449bb3

Do I need to do anything special with fb blanking for the Vero 4K? Right now Moonlight blanks both fb0 and fb1 on Amlogic devices. There are also things like `/sys/class/video/disable_video` (which defaults to 1) that maybe I need to mess with?

If you've got a simple amcodec sample that works on the Vero 4K, I can probably easily mimic what that does. I just couldn't get any of the ones from the old amcodec repo working at all.

cgutman, 2022-11-01:
> As you see, without that map you will never ever get any frame out of the Vero.

On a freshly booted Vero (with Kodi disabled), it looks like vfm is set to `default { decoder(0) ppmgr(0) deinterlace(0) amlvideo(0) amvideo}` which works for us. Is that default initialization safe to rely on?

> You can then decide (e.g. based on that timestamp) when that frame should be displayed. You do that by calling the ReleaseFrame function, which tells the kernel that the given frame should be passed to the next module on the vfm map. The "amvideo" module is then doing all the magic to finally display that frame.

Aha! This was the missing part. Adding a pair of `VIDIOC_DQBUF` and `VIDIOC_QBUF` `ioctl()` calls makes the video display properly.

This code was previously used by other Amlogic decoders that apparently didn't need these ioctls to be invoked to display the output on screen. How do I tell programmatically if I'm dealing with a decoder that needs VIDIOC_DQBUF+VIDIOC_QBUF vs. one where `codec_write()` alone is enough? I don't want to break existing Amlogic devices with the Vero changes.

> Most often the reason is that the encoded frame had some errors in it.

Yep, the reason is obvious in hindsight. `DECODER_TYPE_FRAME_MODE` is telling the decoder we will give it full frames, but we call `codec_write()` multiple times with partial frames. `DECODER_TYPE_SINGLE_MODE` seems to work for me for both H.264 and HEVC without issues.

[59 more replies]