moonlight-stream/moonlight-embedded #574: Moonlight works just 1 or 2 minutes and then disconnects.

Spamm00r, 2017-10-08:
Please provide the following info.

**_NVidia Geforce Experience version: 385,69 GTX 960 Ti NVIDIA GeForce GTX 960, GFE 3.9.0.61 (gs_04_00_22663502, 7.1.400.0)
**Moonlight Embedded version:latest. Updated yesterday via respository. Moonlight Embedded 2.4.3 (EMBEDDED;ALSA;PI)
**Moonlight Embedded source:Updated yesterday via respository.
**Moonlight Embedded running on:** Raspberry Pi 1
**Moonlight Embedded running on distribution:OSMC

**Verbose output `-verbose` of Moonlight Embedded:** ? I dont know what you mean? How can I make these logs?

**What is the expected result? Moonlight schould work as long as I dont disconnect myself.

**What happens instead of that? Monlight works just 1 to 2 minutes and then disconnects without any visible error message. And I have to reconnect again. This problem started occuring only recently. Before that I could use moonlight normally, only issues I had was sometimes bad wifi. I upgraded the wifi to 5gh and now have 120 Mbits wifi and I also use now an stronger Power Supply, with at least 2 Amps.. SO I can rule out bad wifi reception. Also moonlight app on my iphone works flawlessly with my setup.

WHat migth be the issue?

MrCapone, 2017-10-11:
Do you really think Moonlight disconnects? I have similar issue: moonlight works some time then disconnects, but audio and controls continue working, only video turns off. Then I discovered that my board where i have Moonlight turns off video output due to power saving settings.

Spamm00r, 2017-10-15:
Yes moonlight disconnects. It was working flawlessly for over 1,5 years on the same device.
Maybe its and update issue?

This is the osmc log:

https://paste.osmc.tv/upimarejuf

How Can i downgrade moonlight to an older build?

Spamm00r, 2017-10-16:
I cant edit my first post. I figured out how to -verbose. Here is the error output:

Video Ping: send() failed: 101
Stopping input stream...done
Stopping audio stream...Control stream connection failed
Loss Stats: Transaction failed: 101

Spamm00r, 2017-10-16:
Here is the full verbose Log:

Moonlight Embedded 2.4.3 (EMBEDDED;ALSA;PI)
Connect to 192.168.178.100...
NVIDIA GeForce GTX 960, GFE 3.1.2.31 (gs_03_40_21369817, 7.1.340.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 35171, but got 35172)
Control stream connection failed
Loss Stats: Transaction failed: 101
Stopping input stream...done
Stopping audio stream...Video Ping: send() failed: 101
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

Spamm00r, 2017-10-16:
Same problem with lates GFE from today:

Moonlight Embedded 2.4.3 (EMBEDDED;ALSA;PI)
Connect to 192.168.178.100...
NVIDIA GeForce GTX 960, GFE 3.10.0.95 (gs_04_02_22962627, 7.1.402.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 27966, but got 27967)
Control stream connection failed
Loss Stats: Transaction failed: 101
Stopping input stream...done
Stopping audio stream...Video Ping: send() failed: 101
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done