moonlight-stream/moonlight-embedded #576: Exit Streaming Session via controller does not Quit moonlight or launched program on Host

craciant, 2017-10-14:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.9.0.97
**Moonlight Embedded version:** 2.4.1
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result?** Moonlight quits on Play+Back+LeftShoulder+Rightshoulder, closing the stream client on Pi and launched host application on server.

**What happens instead of that?** The client closes, but moonlight does not quit and the launched application continues to run on the host server.

Why is this command's result different than "moonlight quit?"
Same result with Ctrl+Alt+shit+Q

Can these shortcuts be configure to execute the command "moonlight quit?" 

I found this supposed workaround on issue #317  
"What i use for retropie is runcommand for lauching .sh
put all your .sh in rom folder "steam"
create folder inside configs named "steam" and add emulators.cfg , write
steam = "%ROM%"
default = "steam"

on es_systems.cfg add
<system> <fullname>Steam</fullname> <name>Steam</name> <path>~/RetroPie/roms/steam</path> <extension>.sh .SH</extension> <command>/opt/retropie/supplementary/runcommand/runcommand.sh 0 _SYS_ steam %ROM%</command> <platform>steam</platform> <theme>steam</theme> </system>
and simply add to /opt/retropie/configs/all/runcommand-onend.sh and write
moonlight quit IP"
But current version of Retropie does not have a runcommand-onend.sh file that I can find, at least not in the location referenced.

craciant, 2017-11-11:
ok there was a simple solution to this one- by default there is no runcommand-onend.sh file, but looking at runcommand.sh revealed that it does attempt to run that file on exit, so if you create it and put in that directory it will work. 

to future users, I suggest writing some bit of code to detect if you are using a streaming system, or copying runcommand and creating a runcommandstream.sh file to go with a runcommandstream-onend.sh file otherwise retropie will attempt to quit moonlight every time you quit a game, streaming or not.

closed 2017-11-11 by craciant

morfeus77, 2019-12-15:
I have this weird issue.  When I quit the moonlight stream with the shortcut on joypad, it succesfully returns to my emulstation but keeps the game running on my pc.  I created a shell script doing nothing but "moonlight quit" which closes the game on the pc.  I wanted to do this inside /opt/retropie/configs/all/runcommand-onend.sh but this behaves different.  When inside the runcommand-onend.sh the "moonlight quit" command tells me I need be paired first.  Any clues as to why this script is not working properly inside my runcommand-onend.sh (I tried it from bash as well keeps asking to be paired first).