moonlight-stream/moonlight-embedded #577: Sound stutter!

DocMAX, 2017-10-17:
Please provide the following info.

**_NVidia Geforce Experience version:_3.10.0.95**
**Moonlight Embedded version:2.4.3**
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
**Moonlight Embedded running on:Arch Linux** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
**Moonlight Embedded running on distribution:Arch Linux** _Arch Linux/Raspbian/OpenELEC/..._

**Verbose output `-verbose` of Moonlight Embedded:**
Moonlight Embedded 2.4.3 (SDL;X11;VDPAU;VAAPI;EMBEDDED;ALSA;PULSE)
Connect to game-w7-vm...
NVIDIA GeForce GTX 970, GFE 3.9.0.97 (gs_04_00_22663502, 7.1.400.0)
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva info: va_openDriver() returns 0
Platform X Window System (VAAPI)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Add input /dev/input/event11...
Detected Xbox 360 Wireless Receiver (030000005e040000a102000000010000) on /dev/input/event11 as Xbox 360 Wireless Receiver
Stream 1920 x 1080, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
EVIOCGRAB failed with error 19
Removed input device
Returning RTP packet queued for too long
Received OOS audio data (expected 10957, but got 10958)
Alsa error from writei: -32
Decode failed - Invalid data found when processing input
Decode failed - Invalid data found when processing input
Alsa error from writei: -32
Alsa error from writei: -32
Alsa error from writei: -32
Alsa error from writei: -32
Alsa error from writei: -32
Alsa error from writei: -32
Alsa error from writei: -32
Alsa error from writei: -32
Alsa error from writei: -32
Alsa error from writei: -32
Alsa error from writei: -32
Alsa error from writei: -32
Alsa error from writei: -32

**What is the expected result?**

No sound stutter

**What happens instead of that?**

Sound stutter (also with -audio pulse)

Bosken85, 2018-01-27:
I am having the same error and problem. My setup differs though

Haris1977, 2018-02-10:
I get same sound stuttering with my setup too. I went back to rasbian jessie (from Stretch) and it s a little better. There is still problem though. I only use it to stream a specific page (webtv) in my country as i am a legal member of this. I have a rasp pi 1 (model B), with raspbian jessie.