moonlight-stream/moonlight-embedded #583: Error trying to map out the controller

legraphista, 2017-11-04:
Please provide the following info.

**_NVidia Geforce Experience version:_**  3.10.0.95
**Moonlight Embedded version:** Moonlight Embedded 2.4.4
**Moonlight Embedded source:** http://archive.itimmer.nl/raspbian/moonlight/pool/main/m/moonlight-embedded/moonlight-embedded_2.4.4-1jessie1_armhf.deb
**Moonlight Embedded running on:** RetroPi
**Moonlight Embedded running on distribution:** Raspian Jessie

**Verbose output `-verbose` of Moonlight Embedded:**
```
pi@retropie:~ $ moonlight map -input /dev/input/event0 --verbose
Moonlight Embedded 2.4.4 (EMBEDDED;ALSA;PI)
No mapping available for ShanWan Trust Gamepad (03000000bc2000000055000011010000) on /dev/input/event0
*** buffer overflow detected ***: moonlight terminated
Aborted
```
**What is the expected result?**
map my controller?
**What happens instead of that?**
program crashes

lryjnks, 2017-11-04:
I've the same Problem, with my 'Microsoft X-Box 360 pad'
Is there any solution?

legraphista, 2017-11-05:
As a workaround, I reverted to 2.2.0, did the mapping and went back up to 2.4.4

wstephenson, 2017-11-06:
The map command no longer exists, see this earlier issue: https://github.com/irtimmer/moonlight-embedded/issues/511. It seems you have to use evtest to detect button codes from your controller and then add a line for it to your gamecontrollerdb.txt file. I haven't done this myself yet, so can't help you further.

Kazatel, 2017-12-14:
https://github.com/irtimmer/moonlight-embedded/releases
Changelog:
    Add map command for easy creation of mappings

in version 2.4.4 was map command added back, however it is not working, or at least same error for me.

uzumis, 2017-12-29:
I have the same problem. I can`t use my right Analog Stick.. their Y+ Y-.. only X+ X-  
when I remap .. i`fve receive the same msg as OP.

Kazatel, 2018-01-05:
Hi ,

I saw irtimmer post somewhere where he said that retropie use specific
driver and use DPAD mapping in different way, see below pull form retropie
configuration

dpup:*h0up*,dpleft:*h0left*,dpdown:*h0down*,dpright:*h0right*

*So I am facing bigger issue, I guess. *

Did you try to use "sdl2-jstest" to find out exact mapping and then to
update "/usr/share/moonlight/gamecontrollerdb.txt" with someting
appropriete ID and mapping for your gamepad?

0500000049190000020400001b010000,GamepadPlus,a:b0,b:b1,x:b4,y:b3,leftshoulder:b6,lefttrigger:b8,rightshoulder:b7,righttrigger:b9,back:b10,start:b11,leftstick:b13,rightstick:b14,leftx:a0,lefty:a1,rightx:a2,righty:a3,dpup:h0.1,dpleft:h0.8,dpdown:h0.4,dpright:h0.2,platform:Linux,

as you can see in gamecontrollerdb.txt there is no x+ x- , y +,y- etc. but
a1,a2,a3 etc. so make sure you know what axis representing right analog
stick and you have it properly mapped.

Check your moonlight version and update if you can, then try to create
mapping  and pull all info as I did in my post.

Hope you have some luck.

Regards Tomas