moonlight-stream/moonlight-embedded #584: SDL Controller mapping not using gamecontrollerdb.txt

FockeOeuf, 2017-11-07:
Please provide the following info.

**_NVidia Geforce Experience version:3.10.0.95
**Moonlight Embedded version:2.4.3
**Moonlight Embedded source: Coming from Kodi addon
**Moonlight Embedded running on: Intel NUC i3 Haswell
**Moonlight Embedded running on distribution: LibreElec

**Verbose output `-verbose` of Moonlight Embedded:**
/storage/.kodi/addons/script.moonlight/bin/moonlight stream -packetsize 1024 -bitrate 20000 -1080 -fps 60 -surround -nosops -app Steam -keydir /storage/.kodi/userdata/addon_data/script.moonlight/keys -mapping /storage/.config/moonlight/gamecontrollerdb.txt -verbose
Moonlight Embedded 2.4.3 (SDL;ALSA;PULSE;CEC)
Searching for server...
Connect to XXX.XXX.XXX.XXX...
NVIDIA GeForce GTX 1060 6GB, GFE 3.10.0.95 (gs_04_02_22962627, 7.1.402.0)
Platform SDL2 (software decoding)
Stream 1920 x 1080, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Decode failed - Invalid data found when processing input
Decode failed - Invalid data found when processing input
^CStopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed
Loss Stats: Transaction failed: 11
doneStopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
Error message regarding controller not found (as I suppressed all the lines in gamecontrollerdb.txt except a dummy one)

**What happens instead of that?**
No error message and a messy controller mapping:
XBox 360 pad: buttons correctly mapped but triggers stay in push state. When I release the trigger, it's like the "oof" state is not recognized and in the game the behavior keeps being "trigger actioned"
PS3 pad: not even recognized, no input detected, can move in Steam Big Picture

I tried and checked with various GUID, nothing changes. Tried changing the name of the gamecontrollerdb.txt file and I do get a "file not found" error when launching moonlight, so I know the file I modify is clearly searched by the binary, but it's like it doesn't read the content as it seems the default profile is always used, even if I delete the two first lines (default mappings) of the file.

I have the same behavior whether I use the -mapping option or not.

Weird thing is I never get the controller not found mapping error whatever I do, which is not the same issue as most of the people raising controller mapping issues.

Any idea?

FockeOeuf, 2017-11-07:
Ok I've just seen the 2.4.4 patch note regarding triggers (already read that but makes sense now), so only remains the fact that I can't succeed in making anything else than my X360 pad to work with gamecontrollerdb.txt file.

If I only keep the PS3 controller mapping in the file, the X360 pad still works the same way as before, and not the PS3 pad...

pfaffenrodt, 2018-01-26:
Same here with my PS3 controllers. GUID of my controllers are in gamecontrollerdb.txt and still moonlight.

I wish I could use my old mapping file