moonlight-stream/moonlight-embedded #590: Stream crashes and stutters

remingtonH, 2017-11-14:
Please provide the following info.

**_NVidia Geforce Experience version:3.10.0.95_**
**Moonlight Embedded version:2.4.4**
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
**Moonlight Embedded running on:** _Raspberry Pi 3
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._

**Verbose output `-verbose` of Moonlight Embedded:**
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 34925, but got 34926)
Unrecoverable frame 588: 5+0=5 received < 7 needed
Network dropped an entire frame
Waiting for IDR frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Returning RTP packet queued for too long
Received OOS audio data (expected 39175, but got 39177)
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Returning RTP packet queued for too long
Received OOS audio data (expected 39203, but got 39205)
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Returning RTP packet after queue overgrowth
Received OOS audio data (expected 39230, but got 39232)
Returning RTP packet queued for too long
Received OOS audio data (expected 46088, but got 46092)
Returning RTP packet queued for too long
Received OOS audio data (expected 46093, but got 46095)
Returning RTP packet queued for too long
Received OOS audio data (expected 46101, but got 46104)
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
Streaming at a clean and steady rate at 1080
**What happens instead of that?**
the pi streams steam, and it works with relatively no lag for about a minute, then I start getting these errors, and it gets choppy and laggy, and crashes

HexHyte, 2018-03-03:
Same exact problem here.. seems that is a bitrate related problem to me.
Streaming 1080p at 45fps with 15k bitrate everything go amazingly smooth for about a minute and then BAM it starts to lag, freeze and eventually crash... why this is happening??

Droid22, 2018-03-04:
same problem here. quite annoying