moonlight-stream/moonlight-embedded #592: ffmpeg libav version to compile

FockeOeuf, 2017-11-14:
Hi,

I'm trying to compile moonlight with vaapi, and I get an error with the AVCodecContext::hw_device_ctx property which is does not exist in the libavcodec.h I get from the Debian 9 and Ubuntu Xenial ffmpeg packages I get.
Seems that this property does not exist in the branches used for Debian and Ubuntu, but I can't really get which release I need to use in order to be able to compile. Release 2.8 solves the hw_device_ctx issue, but raises another one.

Is it possible to get ffmpeg branch used in moonlight? Or even better have it included in the pkg_check_modules command in the Makefilelist?

Nyaran, 2018-03-23:
I have same problem.
Here is the output:

``
[ 92%] Building C object CMakeFiles/moonlight.dir/src/video/ffmpeg_vaapi.c.o
/home/pi/sources/moonlight-embedded/src/video/ffmpeg_vaapi.c: In function ‘va_get_format’:
/home/pi/sources/moonlight-embedded/src/video/ffmpeg_vaapi.c:51:10: error: ‘AVCodecContext {aka struct AVCodecContext}’ has no member named ‘hw_device_ctx’; did you mean ‘hw_frames_ctx’?
   context->hw_device_ctx = device_ref;
          ^~
``

EdgeSync, 2018-12-23:
I was able to fix this error by making the following change in:

moonlight-embedded/src/video/ffmpeg_vappi.h

Line 51:
change from
context->hw_device_ctx = device_ref;

change to
context->hw_frames_ctx = device_ref;

quannv108, 2019-06-12:
> I was able to fix this error by making the following change in:
> 
> moonlight-embedded/src/video/ffmpeg_vappi.c
> 
> Line 51:
> change from
> context->hw_device_ctx = device_ref;
> 
> change to
> context->hw_frames_ctx = device_ref;

When I change to `hw_frames_ctx`, the code is not running anymore, no frame can decode