moonlight-stream/moonlight-embedded #594: 8bitdo SNES30 Gamepad not in gamecontrollerdb.txt

joshsmithtogo, 2017-11-19:
Please provide the following info.

**_NVidia Geforce Experience version:_**
**Moonlight Embedded version:**
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result?**

**What happens instead of that?**
No mapping available for 8bitdo SNES30 GamePad 05000000c82d0000402800000001000

Haris1977, 2017-11-23:
No mapping available for SPEEDLINK XEOX PRO ANALOG Gamepad (SL-4446-BK-01) 03000000341a00000608000010010000

irtimmer, 2019-04-30:
Currently controller mappings are imported from https://github.com/gabomdq/SDL_GameControllerDB Therefore new controller mappings needs to be submitted to them.

closed 2019-04-30 by irtimmer