moonlight-stream/moonlight-embedded #595: Destiny 2 Can't Launch

Calamity1911, 2017-11-22:
nvidia GeForce Experience version 3.0.11.73 (finally added support for streaming Destiny 2)
latest version of moonlight embedded as of 11/22/2017
Raspberry Pi 3 running Raspbian Stretch 

I installed moonlight embedded just fine and I tried to stream destiny 2, but it couldn't find the game.
I know I am supposed to have logs but I can't get them now, I'm short on time. I typed "moonlight stream [my PC's IP] -list" and Destiny 2 was option 4, so I then went and typed "moonlight stream [my PC's IP] -app Destiny 2 -1080 -fps 30 -bitrate 4000 -nosops" and it said something along the lines of could not find destiny 2. 
Again, I'll come back later with exact logs on what happened. 
I then did those same settings but with "-app Steam" and "-app mstsc.exe" and both worked just fine, so maybe it's the space in Destiny 2's name. Since the -list option displays games with numbers next to them, I then tried "-app 4" and it had the same error as Destiny 2 but the app name was "4" 

Again, I am sorry for the scant details. I will comment the exact logs in an hour or two.

Calamity1911, 2017-11-23:
Alright, I feel dumb. I have the solution. 
put the game name in quotes. 
For Destiny 2:

`moonlight stream [PC IP] -1080 -fps 30 -bitrate 4000 -codec auto -nosops -app "Destiny 2"`

closed 2017-11-23 by Calamity1911

Eschguy, 2018-10-18:
Good god...thank you so much. F'ing quotes...