moonlight-stream/moonlight-embedded #596: Multiple (two) controllers (Xbox One) not appearing in game.

cg-david, 2017-11-23:
Please provide the following info.

**_NVidia Geforce Experience version: 3.11.0.73_**
**Moonlight Embedded version: 2.4.4**
**Moonlight Embedded source:** _ http://archive.itimmer.nl/raspbian/moonlight jessie main_
**Moonlight Embedded running on:** _Raspberry Pi_
**Moonlight Embedded running on distribution:** _RetroPie_

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result?**
When starting `moonlight stream -fps 60 -1080 -bitrate 30000 xxx.xxx.xxx.xxx -app _SoccerGame_`, two controllers will be recognized and two players can play against each other.

**What happens instead of that?**
Only one controller available. 

**Peripherals**
Xbox-One Pad (Controller) x2

Hey. I have set up moonlight-embedded under RetroPie successfully and the system is up and running smoothly. I am still testing on different bitrates, but ok. I know, that some friends will come over on the weekend and I'd love to have two controllers up and running. In the RetroPie menu, both controllers work. 
I have been going through some posts, where people had problems with their controller and needed to map them. However, I don't need to specify any mapping, for one player moonlight will just work fine. My Xbox-One controllers appear in `/usr/share/moonlight/gamecontrollerdb.txt`. I needed to adjust the buttons a little and when starting moonlight, the verbose output will tell me, that some mapping is wrong. I haven't experienced any false mapping so far. Still moonlight will recognize both controllers and give this message twice. However the controllers do not appear in the game. Any idea?

Tamika305, 2018-07-16:
@cg-david , did you manage to get two controllers assigned within moonlight? I'm going to attempt to do the same tonight, so any tips would be useful.

cgutman, 2021-07-26:
Reopen if this still affects v2.5.0

closed 2021-07-26 by cgutman