moonlight-stream/moonlight-embedded #609: New Windows/GFE update breaks Moonlight? (RTSP: Failed to connect to UDP port 48010)

Trinity92, 2018-01-10:
**_NVidia Geforce Experience version:_** 3.12.0.79
**Moonlight Embedded version:** 2.2.4
**Moonlight Embedded source:** _included in Raspbian's package manager_
**Moonlight Embedded running on:** _Raspberry Pi 3_
**Moonlight Embedded running on distribution:** _Raspbian_

**Verbose output `-verbose` of Moonlight Embedded:**
pi@raspberrypi:~ $ moonlight stream -1080 -fps 60 -app "Steam" -platform pi -verbose
Moonlight Embedded 2.4.4 (EMBEDDED;ALSA;PI)
Searching for server...
Connect to 192.168.178.104...
NVIDIA GeForce GTX 750 Ti, GFE 3.12.0.79 (gs_04_04_23287129, 7.1.404.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Detected Logitech Logitech Dual Action (030000006d04000016c2000010010000) on /dev/input/event0 as Logitech Dual Action
Mapping for Logitech Logitech Dual Action (030000006d04000016c2000010010000) on /dev/input/event0 is incorrect
Stream 1920 x 1080, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...RTSP: Failed to connect to UDP port 48010
failed: -1
Cleaning up platform...done
^Cpi@raspberrypi:~ $

**What is the expected result?**
I would like to stream my desktop PC Steam library to my Raspberry Pi 3 through Moonlight.

**What happens instead of that?**
Following a Windows 8.1 and Nvidia GFE update, Moonlight now hangs up at the "Cleaning up platform...done" line and I have to kill it manually with Ctrl+C. The game footage is not streamed to the Raspberry Pi 3 as a result, although I can still see the game's video on my host PC's screen (I can't hear any audio from either device though). The newest version of the Android Moonlight app also ceased working with my setup for the exact same reasons.

The weird thing is that it used to work before (even yesterday afternoon!) on the exact same setup and now, after updating Windows and Nvidia GFE to the newest versions, it suddenly stopped working and always gives me that error.
I followed the troubleshooting guide and tried disabling the Firewall or adding an exception, re-installing Nvidia GFE, but none of these solutions worked. Keep in mind that I am running Moonlight in my local network over a switch, so no port forwarding necessary (even though the ports are in fact forwarded correctly on my router, but that may not be relevant for this issue).

I may try to re-install Moonlight altogether and see if it works, altough I doubt that it might solve the issue. IMO, this could be due to the new updates for GFE and Windows 8.1 somehow breaking the current Moonlight implementations... or perhaps I am overlooking something? Anyone else having this same issue?

goli333, 2018-01-10:
Same issue with new Windows 7 and GFE update... Tried Hamachi, port fowarding and reinstalling but none of them helped.

cgutman, 2018-01-10:
Yes, I have pushed fixes to the moonlight-common-c repository. Once @irtimmer integrates these changes, it will work again.

Trinity92, 2018-01-10:
Thank you very much for your quick fix! I will test it as soon as it has been integrated.

davidoneacre, 2018-01-11:
Can confirm, got the exact same error after updating GFE on Windows 10. Downgraded to GFE 3.10.0.95 and everything works again.

irtimmer, 2018-01-13:
The required changes for GFE 3.12 in moonlight-common-c are now integrated into the new release 2.4.5

closed 2018-01-13 by irtimmer

Silent-Hunter, 2018-02-22:
This seems to be happening again. I'm using Debian 9, moonlight-embedded 2.4.6, and GFE 3.12.0.84.
It might be my computer though, Dukto, a local network file sharing app is also not working. But I can't begin to imagine what the issue could be.

Meems33, 2018-03-22:
Same. GFE 3.13.1.30 moonlight-embedded 2.4.6 on pi 3

Meems33, 2018-03-23:
I redact my statement. After uninstalling and reinstalling moonlight-embedded it works.

cgutman, 2018-03-23:
Maybe stale .so libraries that the DEB didn't overwrite?

Meems33, 2018-03-23:
I suspect that is the case. Either the uninstall + reinstall fixed it or the make install from master fixed it.

Alevale, 2018-06-01:
I can confirm that 3.13 has a similar issue as this (wouldn't let me actually stream the game even locally) but 3.14 fixes the issue in which the computer is not online