moonlight-stream/moonlight-embedded #611: Maybe a decoding problem

k4mik4z, 2018-01-13:
Please provide the following info.

**_NVidia Geforce Experience version:** 3.12.0.79
**Moonlight Embedded version:** 4.4.5
**Moonlight Embedded source:** compiled from rasbian/stretch
**Moonlight Embedded running on:** Raspberry Pi 3 
**Moonlight Embedded running on distribution:** Rasbian 

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result?**

**Moonlight Embedded source: compiled from rasbian/stretch
**Moonlight Embedded running on: Raspberry Pi 3 
**Moonlight Embedded running on distribution: Raspbian stretchmoo

**Verbose output `-verbose` of Moonlight Embedded:**
Moonlight Embedded 2.4.5 (PI;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.1.10...
NVIDIA GeForce GTX 1080, GFE 3.12.0.79 (gs_04_04_23287129, 7.1.404.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/local/share/moonlight/gamecontrollerdb.txt
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Stopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed
Loss Stats: Transaction failed: 11
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
I should have a clear image

**What happens instead of that?**
Instead of that I get this kind of image.
They mouse and keyboard sharing are totally working.
I've tried different options and got this result each time.
[image]

RWoof, 2018-01-14:
I'm having the same issue, I see this grey screen as soon as I connect. If I move the mouse or grab a window and drag it around it kind of erases the grey but not perfectly.

cgutman, 2018-01-14:
My guess is the culprit is https://github.com/moonlight-stream/moonlight-common-c/commit/1c386a898731ae07168d52c835657409b54f7790

After that change, IDR frames come in as one single frame, rather than a "frame" each for SPS, PPS, and picture data. If the Pi decoder assumes that it will receive only a single 4-byte NALU as part of each buffer submitted, this will need a change in the decoder to submit them separately (like https://github.com/moonlight-stream/moonlight-ios/commit/4b9bd8398b0fdd521fecba8f610f312a03c8e328 or https://github.com/moonlight-stream/moonlight-chrome/commit/053d2944d1569bd5ae71bebea0ced8f65d2472d3).

I originally thought the culprit was gs_sps_fix() but other than not properly updating the DECODE_UNIT.fullLength field (which is unused in the Pi decoder), it looks like it should still work after my moonlight-common-c changes because it doesn't assume the SPS is the only buffer in the frame (though it does assume the SPS will always be the first buffer in the IDR frame, which is true). I could be missing a bug there though.

irtimmer, 2018-01-14:
Version 2.4.6 is now submitting the SPS and PPS separately from the picture data and should solve the corruption at the Raspberry Pi. Other decoder are unchanged cause, as far as I known, they don't require changes. Although they are not all tested as I don't have access to all the different decoders supported in Moonlight Embedded.

closed 2018-01-14 by irtimmer