moonlight-stream/moonlight-embedded #613: windows shortcuts not working

saidbet, 2018-01-21:
**NVidia Geforce Experience version: 3.12.0.84
**Moonlight Embedded version: 2.4.6
**Moonlight Embedded source: repository
**Moonlight Embedded running on: Raspberry Pi3
**Moonlight Embedded running on distribution: Raspbian Jessie

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result? Alt+Tab, Alt+F4, Windows key, Alt+Esc should be working ?

**What happens instead of that? Nothing happens

I tried with multiple keyboards and multiple keyboard layouts but can't seem to get windows shortcuts working. The windows key does nothing and alt+... combinations don't seem to work.
I managed to get it working by using VirtualHere, but I can't rely on it because I need to use it at work, and I don't have control over the router to forward ports to my Rpi.

lambia, 2018-02-11:
It seems to be a limitation of Gamestream protocol, or so I heard. But I can't find anything on this topic. Nonetheless I suppose that should be bypassed doing some sort of re-binding of keycodes. Let's wait for cugtman that previously closed a similar [thread ](https://github.com/moonlight-stream/moonlight-android/issues/83#event-676475512)
I also need a solution as hell. Just this fix and moonlight would be THE solution for cloud computing.

lambia, 2018-02-11:
Yep, but using different keys for alt-tabbing (one of the most used shortcuts) or for win-shortcuts (win-r, win-d, win-arrows) is not as comfy and intuitive as using original keys. Now I'd like to understand what is the limitation. From the application to intercept windows shortcut before it does, or from the protocol to send those specific keycodes. I want to know the limitation, cause both of them are solvable.

MakiseKurisu, 2021-02-05:
I think we can revisit this issue since in the latest moonlight-qt the system key capture is added.

cgutman, 2021-07-24:
This is supported now in GFE 3.20 and later.

closed 2021-07-24 by cgutman