moonlight-stream/moonlight-embedded #614: RetroPie, Xbox 360 controller mapping in Steam

smathev, 2018-01-23:
Please provide the following info.

**_NVidia Geforce Experience version:3.10.0.95
**Moonlight Embedded version: 2.4.6
**Moonlight Embedded source: Built via TechWizTime, using apt-get repository
**Moonlight Embedded running on: Raspberry Pi
**Moonlight Embedded running on distribution: RetroPie

**Verbose output `-verbose` of Moonlight Embedded:** 
pi@retropie:~ $ moonlight stream -verbose
Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to 192.168.1.78...
NVIDIA GeForce GTX 960, GFE 3.10.0.95 (gs_04_02_22962627, 7.1.402.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile gamecontrollerdb.txt
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
...

There are no critical errors, steam starts fine and everything seems to work. - but when I start inputting commands on my 360 controller (both in Steam Big Picture) and in a given game (Witcher 3), the inputs don't match the buttons.
Some of the verbose is for instance:
"Unmapped button: 706" (which I believe is when I press the dpad-right or dpad-left)
"Unmapped button: 707"

**What is the expected result? I expected steam to start, and my XBOX 360 controller to map correctly.

**What happens instead of that?** Most mappings are wrong:
DPAD Left/Right is DPAD Up/Down, Lstick up/down is Lstick left/right, etc.

I've identified my DeviceID (00030000045e000002a1000001070000) using /proc/bus/input/devices, but that didn't work. Forcing -map /dev/input/event2 (the eventID of my device) gives me a different ID: 030000005e040000a102000007010000 which didn't work either.

I've used both SDL2-jstest, jstest and evtest to map out the buttons, they all give the same mappings, but entering the complete set into gamecontrollerdb.txt changes nothing no matter which DeviceID I use. I've tried removing all other mappings but still no changes.

/proc/bus/input/devices lists my controller(s) as Xbox 360 Wireless Receiver (XBOX). It's an unofficial receiver.

I've googled everything I could think of, and tried everything all guides says to try - and I've read most subjects in here. The closets issue I could find was: https://github.com/irtimmer/moonlight-embedded/issues/514#issuecomment-310163270 but following the provided solutions has done nothing additional for me.

iDm1, 2018-01-28:
Change /etc/modprobe.d/xpad.conf option triggers_to_buttons to 0, reboot and reconfigure controller button mappings again in Retropie menu.

smathev, 2018-01-30:
@iDm1 Perfect, it seems to have solved my issues... At least some of them, now I have a problem when re-configuring in RetroPie. 

Since it considers my triggers as Axis, it means that when I map triggers in Retropie it fails (somewhat). Left and Right Trigger is now registered as two separate axis, each constantly in the negatives. Depended on how much I press the trigger it moves toward positive. So, when I map the first trigger, it registers Axis+ as the first trigger, and since it is then moving from positive toward negative it registers Axis- as the second trigger.

I haven't tested the mapping in a game-setting yet, as I didn't have time. But I fear that this might cause a new issue. I will return as I start the testing.

iDm1, 2018-01-30:
Why do you consider this as a problem? Triggers are working as intended in emulators with this configuration any way and Moonlight doesn't care about EmulationStation key mapping at all.

smathev, 2018-01-30:
@iDm1 Ok, I didn't know :) I was simply worried about it, cause I couldn't quite map the keys correctly in RetroPie - and wondered if that behaviour was mirrored in the emulators.

But, if everything is working then I was worried about nothing. I will report back, when I've had time to test it.

digix77, 2018-02-10:
My X360 wireless receiver with device id: 030000005e040000a102000000010000 doesn't work properly in Steam if i use the gamecontrollerdb.txt, ex. rightstick is like the guide button and rightx isn't recognized. But if i use the following command to start streaming, everything works fine:
`SDL_GAMECONTROLLERCONFIG="030000005e040000a102000000010000,X360 Wireless Receiver,a:b0,b:b1,y:b3,x:b2,start:b7,guide:b8,back:b6,leftstick:b9,rightstick:b10,leftshoulder:b4,rightshoulder:b5,dpup:b13,dpleft:b11,dpdown:b14,dpright:b12,leftx:a0,lefty:a1,rightx:a3,righty:a4,lefttrigger:a2,righttrigger:a5," moonlight stream -1080 -fps 60`

I also have tried only this device in gamecontrollerdb.txt and it doesn't load the correct mapping.

ottk3, 2018-02-24:
Hi @digix77 ,

I got the same device (030000005e040000a102000000010000) working properly with steam. Check out my [pull request](https://github.com/irtimmer/moonlight-embedded/pull/625) with the mapping.

Make sure your packages are all up-to-date (update all packages from source) and disable/uninstall all additional controller drivers in RetroPie via "Manage packages" --> "driver" --> "Remove all driver packages".

Kr
Sven

david7102, 2018-12-16:
@iDm1 sorry bit of a noob here. I get permission denied

@retropie:~ $ /etc/modprobe.d/xpad.conf option triggers_to_buttons to 0

-bash: /etc/modprobe.d/xpad.conf: Permission denied

cgutman, 2021-07-24:
This is fixed in the latest SDL_GameControllerDB imported today.

closed 2021-07-24 by cgutman