moonlight-stream/moonlight-embedded #616: Gamepad stop working after changing streaming computer

Steuv1871, 2018-02-04:
Hi !

**_NVidia Geforce Experience version:_** v3.8.0.89 (try with the latest 3.10 but had to downgrade cause I was stuck with the UDP port 48010 issue)
**Moonlight Embedded version:** 2.4.3-1jessie1
**Moonlight Embedded source:** apt-get
**Moonlight Embedded running on:** RPi3
**Moonlight Embedded running on distribution:** Retropie

**Verbose output `-verbose` of Moonlight Embedded:**
moonlight stream -720 -fps 30 -verbose -mapping /usr/share/moonlight/gamecontrollerdb.txt 192.168.XX.YY
Moonlight Embedded 2.4.3 (EMBEDDED;ALSA;PI)
Connect to 192.168.XX.YY...
NVIDIA GeForce GTX 1070 Ti, GFE 3.8.0.89 (gs_04_00_22507081, 7.1.400.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Stream 1280 x 720, 30 fps, 5000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Detected Xbox Wireless Controller (050000005e040000fd02000003090000) on /dev/input/event2 as Xbox One Wireless Controller
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done
**What is the expected result?**
Hi ! I've setup a few month back moonlight on my Retropie (great works on this package btw).
It work perfectly with my Xbox Controller connecter on bluetooth to my Pi.
Until last week when I've changed computer (the streaming source).
I've give the new computer the same IP as the previous one (which is no longer connected to the network), installed GFE and pair the device.
I didn't change anything, any file, on the Pi.

**What happens instead of that?**
The streaming work, I see the outpout on the pi, I can navigate, but only the keyboard/mouse plugged on my Pi works.
The Xbox Controller (which works on Retropie menu and game) don't do anything while I'm streaming, I can only use the keyboard.
The verbose output shows that the controller is well detected (I've try an evtest to be sure, my bluetooth controller is the event2).
The controller ID (050000005e040000fd02000003090000) is in the gamecontrollerdb I user.

What did I do wrong ?

nadaso8, 2018-02-17:
I'd really like to figure this out too. I haven't even been able to get my pi to connect to my xbox controller correctly. it just drops the connection throwing an error saying that no service uses the controller. Am I having a Driver issue?

Steuv1871, 2018-04-01:
Hey !
Thanks for your answer.
I've tried to add the -input parameter but it didn't works. I got the following messages  : 
```
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Add input 050000005e040000fd02000003090000...
Failed to open device 050000005e040000fd02000003090000
```

But, I've find a strange solution: If I plug a usb controller (noname SNES controller) and connect the Xbox Controller with bluetooth, it works... Except for 1 button (select/back)...
I keep looking for a right solution.

@nadaso8 : Have you upgrade your Xbox controller firmware ? You just have to connect the gamepad with the cable on a W10 computer (see this video : https://youtu.be/xMk4umveBzU)

nadaso8, 2018-04-01:
@Steuv1871 
I fixed the problem for myself It turns out that when ertm is enabled on the raspberry pi it refuses to connect to the xbox controller wirelessly then things work, though I have noticed some slowdowns when i disable this protocol. I've goten a semi stable stream about 30 miles away on a fairly crowded connection so i'll call things a success.