moonlight-stream/moonlight-embedded #619: Mapping Xbox360 worries L2 and R2

Minux870, 2018-02-12:
I'm on a fork of Recalbox.
Worries with XBOX360 mapping with xboxdrv.enabled = 1 and 
The key mapping is correct but when I play a STEAM game
Ex Ultra SF4, when I press R2 (strong foot) it blocks me the R2 key thereafter. And when I press L2 (the 3 feet) This disables the guide keys + A + B + L2 + R2

**_NVidia Geforce Experience version:_* v3.12.0.84*
**Moonlight Embedded version:*2.4.6*
**Moonlight Embedded source:*??*
**Moonlight Embedded running on:*Other linux device*
**Moonlight Embedded running on distribution:*BATOCERA X64*

**Verbose output `-verbose` of Moonlight Embedded:**
# /usr/bin/moonlight stream -config /recalbox/share/system/configs/moonlight/moonlight.conf -verbose -mapping /usr/share/moonlight/gamecontrollerdb.txt -app Steam
Moonlight Embedded 2.4.6 (CEC;SDL;ALSA)
Searching for server...
Connect to 192.168.1.83...
NVIDIA GeForce GTX 1060 6GB, GFE 3.12.0.84 (gs_04_04_23287129, 7.1.404.0)
Platform SDL2 (software decoding)
Stream 1920 x 1080, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**

Moonlight launches and works, but worries with L2 and R2 keys in the games

**What happens instead of that?**

Minux870, 2018-02-23:
Nobody to help me with my problem

ottk3, 2018-02-24:
Hello @Minux870,

that might be an interference between different drivers. Normally your moonlight output should give a line "Controller detected", are you sure it was connected properly? Can you paste the output of *lsusb*?

On Xbox 360 Controllers the triggers are sometimes detected as buttons instead of axis.

I assume you use RetroPie? 
Try to update your system and uninstall all unnecessary joystick/controller drivers. The easiest was is via the RetroPie Setup: "Manage Packages" --> "driver" --> "Remove all driver packages". Don't forget to reboot the system afterwards.

Kr Sven

Minux870, 2018-02-26:
Thank you for message
No, I'm on a fork of Recalbox on PCx64. (Batocera)
This is lsusb
[https://pastebin.com/zp4p8eh6](url)

ottk3, 2018-02-26:
Exactly that controller *(idProduct: 0x0719)* works properly on my device with RetroPie and Steam.

You might notice that Emulationstation has falsely detected your triggers as a **type="button"**, but it should be an **"axis"** instead. You can check that with:
`cat /home/pi/.emulationstation/es_input.cfg | grep "trigger"`

1.) Use this mapping in your gamecontrollerdb.txt: 
`030000005e040000a102000000010000,X360 Wireless Controller,a:b0,b:b1,back:b6,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,platform:Linux,`

2.) Update your system:
`sudo apt-get update && sudo apt-get upgrade`

3.) Remove the *xpad* and *xboxdrv* packages:
`sudo apt-get remove xpad xboxdrv`

4.) Reboot your system

Regards
Sven

Minux870, 2018-03-01:
It's exactly the same when I disable xbox drivers