moonlight-stream/moonlight-embedded #620: Cant get stream to appear.

nadaso8, 2018-02-15:
Please provide the following info.
**_NVidia Geforce Experience version:_**
3.12.0.84

**Moonlight Embedded version:**
2.4.4

**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
?? I just recently downloaded it using the instructions under raspbian osmc

**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
raspberry pi 3 with the screen outputting to the Pimeroni Hyperpixle Dpi touchscreen 

**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._
raspbian

**Verbose output `-verbose` of Moonlight Embedded:**
Searching for server...
Connect to xxx.xxx.x.xx...
No mapping available for Touchscreen (03000000001000000001000000001000000001) on /dev/input/event0
Initializing platform...done 
Resolving host name...done
Starting RTSP handshake... RTSP: Failed to connect to UDP port 48010 
failed: -1 
Cleaning up platform...done

**What is the expected result?**
moonlight outputs to the small dpi screen I have installed 

**What happens instead of that?**
moonlight outputs to nothing, and I just see X(the desktop). 
I'm guessing moonlight is trying to force output to Hdmi even though I have the output set to the dpi screen in config.txt . Just wondering if anyone had any guesses at fixing this, input would be greatly appreciated.

nadaso8, 2018-02-15:
the issue is also likely not helped by the failure of 48010, but this should in my past experiences with other branches just cause the stream to shut down after about 30 seconds or so and can be fixed by updating.

irtimmer, 2018-02-15:
GFE 3.12 is not supported by Moonlight 2.4.4

closed 2018-02-15 by irtimmer