moonlight-stream/moonlight-embedded #623: Cant connect to server

ben21684, 2018-02-21:
Please provide the following info.

**_NVidia Geforce Experience version: 3.12.0.84
**Moonlight Embedded version: 2.4.6
**Moonlight Embedded source:** http://archive.itimmer.nl/raspbian/moonlight/ jessie/main moonlight-embedded armhf 2.4.6-1jessie1
**Moonlight Embedded running on:** _Raspberry Pi
**Moonlight Embedded running on distribution:** Raspbian

**Verbose output `-verbose` of Moonlight Embedded: 
pi@raspi:~ $ moonlight pair -verbose
Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to 192.168.216.104...
Can't connect to server 192.168.216.104

**What is the expected result?**
connecting to server, to input pin

**What happens instead of that?**
nothing.

Hi. I was using moonlight(2.4.4) some months ago with this raspi, but I don't know with wich version of GF Exp. So I assumed that there was an update of GFE to 3.12.0.84 so I updated moonlight with apt-get to version 2.4.6
Also I had changed the IP dresses of my PC and Raspi, but both can ping each other.
I also cleaned the cache directory:
pi@raspi:~ $ cd ~/.cache/moonlight
pi@raspi:~/.cache/moonlight $ ls
client.p12  client.pem  key.pem  uniqueid.dat
pi@raspi:~/.cache/moonlight $ rm client.p12 
pi@raspi:~/.cache/moonlight $ rm client.pem 
pi@raspi:~/.cache/moonlight $ rm key.pem 
pi@raspi:~/.cache/moonlight $ rm uniqueid.dat 

Any idea, why I can't connect?

greetings Ben

robbyb20, 2018-02-22:
Ben - I am having this same issue.  I had moonlight working for months.  Then, last night it stopped.

Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to 10.1.1.4...
Can't connect to server 10.1.1.4

geforce drivers : 390.77
GFE : 3.12.0.84
Moonlight : Moonlight Embeded 2.4.6

TyanColte, 2018-02-26:
I'm having the same trouble, although mine connects but then automatically disconnects throwing some error or saying that the stream clean disconnected when i clearly didn't disconnect it. I think GFE changed something in the GameStream protocol to intentionally break open source things like moonlight. i haven't tried moonlight on my android device to see if i get the same result or not but i'd bet I would. I hope somebody can get on this soon and get it fixed to work with the new protocol updates.

DariusQuintus, 2018-04-07:
Same for me. Last week everything was okay. Then, all of a sudden I got "Can't connect to server ..."
I've GFE 3.13.1.30 and Moonlight Embedded 2.4.6 and I'm pretty sure I haven't changed A THING!

Maybe there was some hidden NVidia Update, hell knows.

So, is there any chance of debugging it, like testing if NVidia Shield responds to Moonlight. I mean "Can't connect to server" is some kind of a useless statement in this case :(