moonlight-stream/moonlight-embedded #630: moonlight embedded hangs on searching for server

nadaso8, 2018-02-27:
**_NVidia Geforce Experience version:_**
3.12

**Moonlight Embedded version:**
2.4.6

**Moonlight Embedded source:**
Repository
**Moonlight Embedded running on:**
Pi 3 

**Moonlight Embedded running on distribution:** 
Raspbian Jessie 

**Verbose output `-verbose` of Moonlight Embedded:**
Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
Searching for server...

**What is the expected result?**
moonlight finds my computer and stream starts as normal.

**What happens instead of that?**
Starting this morning (after i'd connected on my local network the previous day) my pi has been unable to find the server unless I specifically point it to the global address. This isn't much of a hinderance as i've modified all of my desktop shortcuts to account for the problem. 
    Though it does worry me because I don't have a static ip, meaning that at the moment if my ISP decides to change my global IPv4 address I will be unable to connect without doing a lookup of my global address.

ottk3, 2018-02-28:
Hi @nadaso8 ,

I'm not sure if I understand your problem. Are the PI and the computer you trying to connect on the same LAN network or are you trying to use moonlight over the internet?

My suggestion is that your PC gets it's IP by your router (via DHCP) and that might change. Since moonlight can't know the new IP address you got these options.
1.) set a static IP address on your PC
2.) set a *static lease* for your PC in the router backend (if supported)

But this is not a moonlight related problem imho.

nadaso8, 2018-03-29:
I had to re enter my ip a couple of times and things were fixed. I never found out what exactly caused it in the end.

closed 2018-03-29 by nadaso8