moonlight-stream/moonlight-embedded #632: Redirect HDMI audio

Gashog, 2018-02-28:
Hello everyone

**NVidia Geforce Experience version:** 3.12.0.84 (latest)
**Moonlight Embedded version:** 2.4.6
**Moonlight Embedded source:** deb http://archive.itimmer.nl/raspbian/moonlight stretch main
**Moonlight Embedded running on:** Raspberry Pi
**Moonlight Embedded running on distribution:** Raspbian

**Verbose output `-verbose` of Moonlight Embedded:**
Some ALSA problems if I type "moonlight stream -audio X7"
(X7 is the name of my bluetooth headphones)

**What is the expected result?**
It has been for several days that I'm searching how to redirect the audio from HDMI speakers to my **bluetooth** headphones. I paired my heaphones and I selected them, and besides moonlight, the audio of all the applications on the raspberry pi goes on the headphones. 
When using the optionof moonlight "-audio", what argument do I need to type if I want to have the sound on my headphones ?

**What happens instead of that?**
I found out that : When I select for exemple the analog output and I type "moonlight stream -audio hw:0,0" the sound goes on the analog port. However, without the audio option even if the analog output is selected, the audio from moonlight  go on the HDMI speakers. 
I read that the problem might come from ALSA, but I struggle to undertsand this.
I also saw that the use of pulseaudio may be usefull.

When I type "aplay -l" to list my audio devices, my bluetooth headphones don't appear.

I hope I made myself clear ! Thanks in advance ! :)

ngalluzzo, 2018-03-27:
Did you have any luck with this? Bluetooth audio works fine for me system-wide but no matter what -audio option I use I can't get moonlight audio to route to bluetooth.

Demeryus, 2018-07-09:
I have exactly the same problem. I tried various commands with -audio but with no luck.