moonlight-stream/moonlight-embedded #637: Mouse doesn't ungrab after pressing ctrl-shift-alt.

StevenTheEVILZ, 2018-03-09:
Please provide the following info.

**Nvidia Geforce Experience version**: 3.12.0.84
**Moonlight Embedded version**: 2.4.6
**Moonlight Embedded source**: AUR package moonlight-embedded
**Moonlight Embedded running on**: Desktop (Ryzen 1800X, GTX 960)
**Moonlight Embedded running on distribution**: Arch Linux

**Verbose output `-verbose` of Moonlight Embedded:**
Note this includes the part where i exit the program.
`Moonlight Embedded 2.4.6 (SDL;X11;VDPAU;VAAPI;ALSA;PULSE;EMBEDDED)
Connect to 192.168.100.2...
NVIDIA GeForce GTX 1060 3GB, GFE 3.12.0.84 (gs_04_04_23287129, 7.1.404.0)
Platform X Window System (VAAPI)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
No mapping available for Power Button (19000000000000000100000000000000) on /dev/input/event1
No mapping available for Power Button (19000000000000000100000000000000) on /dev/input/event0
No mapping available for HDA NVidia HDMI/DP,pcm=3 (00000000000000000000000000000000) on /dev/input/event15
No mapping available for HDA NVidia HDMI/DP,pcm=7 (00000000000000000000000000000000) on /dev/input/event16
No mapping available for HDA NVidia HDMI/DP,pcm=8 (00000000000000000000000000000000) on /dev/input/event17
No mapping available for HDA NVidia HDMI/DP,pcm=9 (00000000000000000000000000000000) on /dev/input/event18
No mapping available for HD-Audio Generic Line Out Front (00000000000000000000000000000000) on /dev/input/event10
No mapping available for HD-Audio Generic Line Out Surround (00000000000000000000000000000000) on /dev/input/event11
No mapping available for HD-Audio Generic Line Out CLFE (00000000000000000000000000000000) on /dev/input/event12
No mapping available for HD-Audio Generic Line Out Side (00000000000000000000000000000000) on /dev/input/event13
No mapping available for HD-Audio Generic Front Headphone (00000000000000000000000000000000) on /dev/input/event14
No mapping available for HD-Audio Generic Front Mic (00000000000000000000000000000000) on /dev/input/event7
No mapping available for HD-Audio Generic Rear Mic (00000000000000000000000000000000) on /dev/input/event8
No mapping available for HD-Audio Generic Line (00000000000000000000000000000000) on /dev/input/event9
No mapping available for PC Speaker (100000001f0000000100000000010000) on /dev/input/event6
Stream 2560 x 1410, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Stopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed
Loss Stats: Transaction failed: 11
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done
`

**What is the expected result?**
When I press ctrl-shift-alt, moonlight-embedded ungrabs the mouse
**What happens instead of that?**
Nothing

cooleyenet, 2018-03-15:
I experience the same thing.

StevenTheEVILZ, 2018-03-16:
I've been mess around with the code and have located the issue. I'll submit a PR sometime within the next week(hopefully tommorow) @cooleyenet.