moonlight-stream/moonlight-embedded #644: Moonlight gamepad guid differs from SDL2 and Recalbox guid

nwildner, 2018-03-20:
**_NVidia Geforce Experience version:_** 3.12.0.84
**Moonlight Embedded version:** 2.4.6
**Moonlight Embedded source:** distribution built-in with buildroot
**Moonlight Embedded running on:** Raspberry Pi 3
**Moonlight Embedded running on distribution:** Recalbox 18.03.16

Hi. Moonlight uses a different guid than SDL2 and Recalbox EmulationStation. If i use `sdl2-jtest` to check the controller ID, i get the following string:

    # sdl2-jstest -l
    Found 1 joystick(s)
    
    Joystick Name:     'PLAYSTATION(R)3 Controller'
    Joystick Path:     '/dev/input/event0'
    Joystick GUID:     05000000504c415953544154494f4e00
    Joystick Number:    0
    Number of Axes:     4

Same way the `gamecontrollerdb.txt` that Recalbox creates before launching Moonlight:

    05000000504c415953544154494f4e00,PLAYSTATION(R)3 Controller,platform:Linux,b:b13,righty:a3,rightx:a2,a:b14,x:b15,righttrigger:b9,rightstick:b2,leftshoulder:b10,dpup:b4,start:b3,dpdown:b6,rightshoulder:b11,dpright:b5,lefttrigger:b8,leftstick:b1,guide:b16,y:b12,lefty:a1,leftx:a0,back:b0,dpleft:b7,# 

However, Moonlight seems to be waiting for `050000004c0500006802000000000000` guid for this controller.

**Then, it comes my request:** Could Moonlight use the same method to calculate GUIDs than SDL2, or use SDL2 built-in libraries to retrieve the gamepad GUID? This should create better integration between all emulation boxes/solutions and Moonlight.

External discussion on Recalbox Forum: [18.03.16 - Moonlight](https://forum.recalbox.com/topic/13261/18-03-16-moonlight)

irtimmer, 2018-03-20:
It looks like the GUID is computed differently in case the controller has no vendor or product id. In these cases the first 11 characters (PLAYSTATION in your case) are copied from the name to the GUID. This have been fixed with commit dcda1a5d133007e94afdb289bb2f850083832e70, although I haven't test it as I don't own such a device.

closed 2018-03-20 by irtimmer

nwildner, 2018-03-21:
Thanks for the feedback @irtimmer . I'll test it as soon as Recalbox updates moonlight-embedded to a version tag that has this commit :)

substring, 2018-03-22:
Confirmed working and solving the issue ! Thank you @irtimmer