moonlight-stream/moonlight-embedded #645: 3 seconds video lag on Raspberry Pi latest firmwares

Astral91, 2018-03-24:
Please provide the following info.

NVidia Geforce Experience version: 3.13.1.30
Moonlight Embedded version: 2.46
Moonlight Embedded source: https://archive.itimmer.nl/raspbian/moonlight stretch
Moonlight Embedded running on: Raspberry Pi 2
Moonlight Embedded running on distribution: Raspbian stretch

Verbose output `-verbose` of Moonlight Embedded: irrelevant

What is the expected result?
Audio and video should be in sync

What happens instead of that?
Video lags behind audio and input between 2 and 3 seconds, making Moonlight unusable.

I can restore functionality by rollbacking to an earlier firmware (5 March works fine for me). I use rpi-update to update/rollback
(https://github.com/Hexxeh/rpi-firmware/commits/master)

Steps to reproduce:
sudo rpi-update

Steps to fix issue:
sudo rpi-update 2659c9e

cgutman, 2018-03-24:
Can you determine the exact kernel that broke it and report it upstream to https://github.com/raspberrypi/linux/issues

Astral91, 2018-03-25:
OK I did the research and it's not kernel related.
The last confirmed commit which works with Moonlight is e0db6ba (https://github.com/Hexxeh/rpi-firmware/commit/e0db6ba2e4a469d062f00cd556dd506c436e8fba)

The issue was introduced in ed6ece6b (https://github.com/Hexxeh/rpi-firmware/commit/ed6ece6b5c1c0d8aceb86568ffe3033bb9ae9ecc)

Both releases are using kernel 4.14.24-v7+. If I'm to trust the release notes, there are only clock/voltage modifications.  

I'll take your hint and report it where you advised.

Astral91, 2018-05-12:
Hi,

I can confirm what knives49 said. To this day I still have the issue with firmware later than e0db6ba, only with 1080p@60 fps.

1080p@30fps and 720p run fine on the latest firmware.
1080p@60fps runs fine on 60db6ba and older.

I tried adding video memory and playing with bitrates to no avail (was running with 256MB which should be more than enough, I still lag with 384MB)

posti85, 2018-05-27:
I've done some research. My configuration is:

NVidia Geforce Experience 3.14.0.139 (GTX 970)
Raspberry Pi 2 b+
Linux kernel 4.14.34-1-osmc
moonlight-embedded 2.4.6-1 stretch

~2 sec lag with 1080 at 60fps
No lag with 720 (30 or 60fps), even 1080 at 30fps

So I play 1080 at 30fps util the (confirmed) problem is fixed!

Astral91, 2018-06-16:
Well... It seems that is was already an old issue in the end : I applied https://github.com/irtimmer/moonlight-embedded/issues/398
and the lag is gone even on the latest firmware. I don't know why the issue appeared in March 2018 for me and 2 years ago for others, but the bottom line is that the issue lies with the RPi2 and not moonlight.

closed 2018-06-16 by Astral91

posti85, 2018-06-16:
Thank you for sharing the workarround but I would prefer playing with 30fps rather than overclocking my Raspberry.

posti85, 2018-06-21:
Thank you for the suggestion :) Anyway, I would keep my Raspberry updated. I've no problem playing at 30fps at all (also, I don't usually play...).

posti85, 2018-06-22:
> I don't know why the issue appeared in March 2018 for me and 2 years ago for others

I don't know if it's your case, but I use OSMC and the kernel 4.14 wasn't introduced there until [april update](https://osmc.tv/2018/04/osmcs-april-update-is-here-2/).