moonlight-stream/moonlight-embedded #646: Bad performance on Raspberry Pi 3 B+ (Use multiple cores?)

kroegerama, 2018-03-27:
Are there any plans on enabling multicore support? On my Raspberry Pi 3 B+ I get 100% on one core and the other ones are on idle at 0-1%. I think, the video stream should be handled on a different core than e.g. the input stream. Also, if hardware decoding is used, no core should be maxed out at 100%.

I'm using moonlight-embedded 2.4.6 on Raspbian Stretch. Both, my PC and the Raspberry are connected to the same 1Gbit-Switch.

I get really bad performance (see log at the bottom) using the default settings (10Mbit) `moonlight stream -1080 -codec h265 -fps 30` or `moonlight stream -1080 -codec h264 -fps 30`.
Reducing the bitrate to ~5Mbit fixes the problem, but the Pi3B+ should be able to handle 300Mbit network traffic just fine.

Another problem is that the input stream starts lagging after a few minutes. Button presses are delayed up to 30 seconds.

```
Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to 192.168.178.26...
NVIDIA GeForce GTX 660, GFE 3.12.0.84 (gs_04_04_23287129, 7.1.404.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Detected Wireless Controller (050000004c050000cc09000000010000) on /dev/input/event0 as PS4 Controller
Stream 1920 x 1080, 30 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Error: cannot keep up
Unrecoverable frame 1: 29+5=34 received < 52 needed
Unrecoverable frame 2: 30+0=30 received < 40 needed
Network dropped an entire frame

[...]

Unrecoverable frame 83: 30+0=30 received < 39 needed
Unrecoverable frame 84: 30+0=30 received < 39 needed
Unrecoverable frame 85: 30+0=30 received < 39 needed
Unrecoverable frame 86: 30+0=30 received < 37 needed
Unrecoverable frame 87: 30+0=30 received < 41 needed
Unrecoverable frame 88: 30+0=30 received < 39 needed
Unrecoverable frame 89: 29+0=29 received < 38 needed
Unrecoverable frame 90: 30+0=30 received < 39 needed
Unrecoverable frame 91: 30+0=30 received < 38 needed
Returning RTP packet queued for too long
Received OOS audio data (expected 55624, but got 55625)
Unrecoverable frame 92: 30+0=30 received < 37 needed
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Unrecoverable frame 96: 30+0=30 received < 40 needed
Unrecoverable frame 97: 30+0=30 received < 40 needed
Unrecoverable frame 98: 30+0=30 received < 39 needed
Unrecoverable frame 99: 29+0=29 received < 38 needed
Unrecoverable frame 100: 29+0=29 received < 37 needed
Unrecoverable frame 101: 29+0=29 received < 38 needed
Unrecoverable frame 102: 30+0=30 received < 32 needed
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Unrecoverable frame 106: 30+0=30 received < 40 needed
Unrecoverable frame 107: 30+0=30 received < 40 needed
Unrecoverable frame 108: 29+0=29 received < 39 needed
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Returning RTP packet queued for too long
```

nadaso8, 2018-03-29:
have you tried running at 720p I use a pi 3 and limit the bitrate to 3000 and that gives me pretty stable results with not too many dropped frames even when i'm in 30 miles away, though if i'm in the local coffee shop it has some trouble with dropped frames.

steventhegeek, 2018-03-29:
I'm using a 3 (non Plus) at the office:  `stream -1080 -codec h264 -bitrate 3500` when my network is busy or `stream -1080` when just me in. I get no increase in input lag over time even on wifi. Home network (GFE) has 20mbit up and office network (Pi) is 100mbit down. I only pass keyboard and mouse over moonlight though, I send my USB hub with Xbox Controller/Warthog over TCP/IP on its own thread.

sanakess, 2022-08-03:
I have the same problem with 1080p60fps and 30fps always have huge 1-2s lags and jitters on my 3b+. Solved by overclock 

`sudo nano /boot/config.txt

arm_freq=1372
over_voltage=5
gpu_freq=588
 
# sdram overclock
sdram_freq=588
sdram_schmoo=0x02000020
over_voltage_sdram_p=6
over_voltage_sdram_i=4
over_voltage_sdram_c=4`