moonlight-stream/moonlight-embedded #647: Hot to go about mapping the pimeroni hyperpixel touchscreen

nadaso8, 2018-03-30:
Please provide the following info.

**_NVidia Geforce Experience version:_**
3.12.0.84
**Moonlight Embedded version:**
2.4.6
**Moonlight Embedded source:** 
installed from instructions on main GitHub page
**Moonlight Embedded running on:** 
Raspberry pi 3
**Moonlight Embedded running on distribution:** 
Raspbian
**Verbose output `-verbose` of Moonlight Embedded:**
Moonlight Embedded 2.4.6 (pI;ALSA;EMBEDDED)
Searching for server ...
Connect to 192.168.1.74...
NVIDIA GeForce GTX 1060 3GB, GFE 3.12.0.84 (gs_04_04_23287129, 7.1.404.0)
Platform Raspberry Pi (Broadcom)
Loading making file /usr/share/moonlight/gamecontrollerdb.txt
Can't map (mousex:x1)
Can't map (mousey:y1)

**What is the hoped for result?**
 the touchscreen works and inputs are streamed to host pc
**What happens instead of that?**
can't map the hyperpixel touchscreen. I'm not really surprised I just don't know the things to type into gamecontrollerdb.txt so getting info on what to enter in there would be nice. as you can see above I've entered in mousex:x1 mousey:y1 as a shot in there dark.

nadaso8, 2018-03-31:
**update** 
when I execute : moonlight map -input /dev/input/event0 -debug
moonlight returns : Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
No mapping available for Touchscreen (03000000010000000100000001000000) on /dev/input/event0
*** buffer overflow detected ***: moonlight terminated
Aborted

dead, 2018-03-31:
Moonlight-embedding does not have on-screen inputs, so you would not be able to play using the touchscreen...

nadaso8, 2018-03-31:
Ok that makes sense, but this touchscreen has drivers attached to it that are able to interact with the mouse in X so I would think it's at least possible to pipe those inputs(mouse x&y along with left click) to moonlight for some basic  touchscreen functionality.