moonlight-stream/moonlight-embedded #649: Bitrate appears to be limited around 10 Mbit/s on Raspberry Pi

sybrnorox, 2018-04-07:
**NVidia Geforce Experience version:** 
3.13.1.30
**Moonlight Embedded version:**
2.4.6
**Moonlight Embedded source:** 
repository
**Moonlight Embedded running on:**
Raspberry Pi
**Moonlight Embedded running on distribution:**
Raspbian

**Verbose output `-verbose` of Moonlight Embedded:**

 moonlight stream 192.168.178.79 -verbose -bitrate 60000

Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
Connect to 192.168.178.79...
NVIDIA GeForce GTX 660 Ti, GFE 3.13.1.30 (gs_04_05_23698325, 7.1.405.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Stream 1280 x **720**, 60 fps, 60000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done

**What is the expected result?**
I expect a stream running at a bitrate of 60 Mbit/s that shows significantly less compression artifacts than running at the default 10 Mbit/s.

**What happens instead of that?**
There is no visible increase in image quality compared to running at a bitrate of 10 Mbit/s and image quality at 10 Mbit/s (and at 60 Mbit/s) leaves quite a lot to be desired. Especially soft color gradients are turned into very rough stepped ones. 
Mind you the stream runs well no matter what the settings, there is no input lag and the stream never drops or anything. Only image quality...well...kinda sucks, no matter what I do. It seems like it is limited around 10 Mbit/s (decreasing further does decrease image quality). 
However I can find no explanation for this limit. I am expecting the answer to be something like "your graphics card does not support higher bitrates" or some such, but I haven't found anything, anywhere giving me a hint as to what's the cause.

toniprada, 2018-08-13:
Same thing happens to me.

MopheusDG, 2018-12-05:
Same for me. RPi3 B+ is not working as it was on my RPi3 B , low quality now matter what bitrate.

kurac338, 2018-12-18:
See my link here: https://github.com/irtimmer/moonlight-embedded/issues/711#issuecomment-448174490

It is quite possible that pushing your Pi above 40 000, simply reverts the setting to default 10 000, as I doubt that it can handle such high bitrate even at lower resolutions (720p). I think that 40 000 is an upper limit for the Pi 3 at 1080p60. Also the GTX 660 Ti could be a weak link in the chain here also. I had a GTX 660 at the time of my LG TV, and when I changed to a GTX 1070 my Moonlight experience improved a lot. It was way more stable. No more connection / stream drops, smoother experience overall on the stream client (RPi). Playing on my server / host without a stream was never an issue.

Personally at 30 000 on my old 39" 1080p LG TV I could not notice the artifacts. Now on my newer 49" Samsung 2160p TV, I could, so I bumped it to 40 000.

If you slam your head at the screen you could still see some artifacts, but sitting at a reasonable distance (as you should) from the TV, it should not be noticeable. (In the near future) you will never get pixel perfect mirror image, as you see on your PC, but with a reasonable tolerance, and distance from the TV it can be good enough.

greenimpala, 2018-12-24:
> It is quite possible that pushing your Pi above 40 000, simply reverts the setting to default 10 000

Interesting - seeing the same effect too. What mechanism would be reverting the bitrate and can it be overridden?

Lukeazade, 2019-02-19:
Did you ever find a fix for this?