moonlight-stream/moonlight-embedded #652: Keyboard shortcut to directly quit streaming from Desktop mode?

Haris1977, 2018-04-13:
Please provide the following info.

**_NVidia Geforce Experience version: 3.13.1.30
**Moonlight Embedded version: 2.4.6
**Moonlight Embedded source: repository
**Moonlight Embedded running on: Raspberry Pi 1
**Moonlight Embedded running on distribution: Raspbian

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result?**

**What happens instead of that?**

I always boot directly into desktop mode (moonlight stream -app mstsc.exe 192.168.X.XXX -1080).
Is there any **keyboard shortcut to directly quit/exit** "streaming session" from within desktop? Ctrl+Alt+Shift+Q does nothing. Thanks!

posti85, 2018-08-01:
Same problem here.

Haris1977, 2018-08-01:
I just put my pc in hibernate mode (start menu->hibernate) thus making raspberry to quit into shell.

kurac338, 2018-08-21:
If my instance freezes, or I need to force quit from the "server" side, I just kill the nvidia network container in the task manager. I found this to be the quickest way, but its a dirty shutdown.

closed 2019-01-24 by irtimmer