moonlight-stream/moonlight-embedded #653: Silent error when loading libmoonlight-aml.so + making it work on Amlogic S905

adriancuzman, 2018-04-13:
Please provide the following info.
Running moonlight embedded on an Amlogic system with LibreElec (S905 in this case) the libmoonlight-aml.so is not loaded because libamcodec si not complete in LibreElec. I've added a print with dlerror and it returns: "usr/lib/libamcodec.so: undefined symbol: audio_decode_basic_init" - "audio_decode_basic_init" is in libamadec.so I think commit https://github.com/irtimmer/moonlight-embedded/commit/f7fadbd9cc6f3c7265a6c497cbd74fb7f7b7f574 should be reverted. It will compile, but it will not work, because there won't be any valid platform returned.

I've tried to use RTLD_LAZY instead of RTLD_NOW , it will load, but the video will not work.
From platform.c:
 void *handle = dlopen("libmoonlight-aml.so", RTLD_NOW | RTLD_GLOBAL);

I've added the missing libraries to complete libamcodec, but the video flickers a lot + there is no sound. I've attached a log with debug, but I see that the codec is reset a lot? Any ideas/pointers on how to fix this?

(by the way, I also have build a Rsp3 image from the same code and moonlight works with my PC system/Nvidia Experience version, so the only problems are Amlogic specific)

OPEN es hevc
Requesting IDR frame on behalf of DR
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
codec_write error: ffffffff
[codec_close_cntl]video codec close return=0!
[codec_reset:1072]ret=0
codec_write error: ffffffff
[codec_close_cntl]video codec close return=0!
[codec_reset:1072]ret=0
codec_write error: ffffffff
[codec_close_cntl]video codec close return=0!
[codec_reset:1072]ret=0
codec_write error: ffffffff
[codec_close_cntl]video codec close return=0!
[codec_reset:1072]ret=0
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
OPEN es hevc
Requesting IDR frame on behalf of DR
Waiting for IDR frame
Waiting for IDR frame
Unrecoverable frame 881: 28+0=28 received < 62 needed
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
OPEN es hevc 

[11_Addons.log](https://github.com/irtimmer/moonlight-embedded/files/1906717/11_Addons.log)

lumilirion, 2018-08-04:
Confirm. Reverting this commit makes possible to run -platform aml

irtimmer, 2018-08-04:
Fault commit has been reverted by a9f85e9

_note: I don't own a Amlogic system and therefore changes for this system are untested_

closed 2018-08-04 by irtimmer

zylo117, 2022-10-02:
same issue on khadas vim3, the frame will flicker when the content of the frame changes a lot. I found out that codec will reset when returned code is -1. If the content suddenly changes to much, the returned code of codec_write will be -1 and trigger codec_reset. The only problem is why the returned code of codec_write is sometimes -1
https://github.com/moonlight-stream/moonlight-embedded/blob/7a32234c0b2fd7603a2fe93640209c3a5bf98495/src/video/aml.c#L97-L104