moonlight-stream/moonlight-embedded #654: [Feature] moonlight-embedded for Rock64

Sk1pp3r, 2018-04-13:
I'd like to request a moonlight-embedded version for [Rock64](https://www.pine64.org/?page_id=7147)  

**Moonlight Embedded running on:** Rock64
**Moonlight Embedded running on distribution:** 2 of them: Debian stretch and Ubuntu Mate. 

Thank you.

irtimmer, 2018-08-12:
As stated on https://github.com/irtimmer/moonlight-embedded/wiki/Compilation you also need for compilation and development:

- libssl-dev
- libopus-dev
- libasound2-dev
- libudev-dev
- libavahi-client-dev
- libcurl4-openssl-dev
- libevdev-dev
- libexpat1-dev
- libpulse-dev
- uuid-dev
- libenet-dev
- cmake
- gcc
- g++
- rbp-userland-dev-osmc (Raspberry Pi and OSMC)
- libraspberrypi-dev (Raspberry Pi and Raspbian)
- fakeroot (Package creation)
- debhelper (Package creation)

closed 2018-08-12 by irtimmer

Sk1pp3r, 2018-08-29:
I've installed the packages, but cmake still shows errors:

rock64@rock64:~/Software/moonlight-embedded/build# cmake ../
-- Could NOT find Broadcom (missing:  BROADCOM_INCLUDE_DIR VCOS_LIBRARY VCHIQ_LIBRARY OPENMAXIL_LIBRARY BCM_HOST_LIBRARY)
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Could NOT find Amlogic (missing:  AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY)
-- Could NOT find Rockchip (missing:  ROCKCHIP_INCLUDE_DIR ROCKCHIP_LIBRARY)
-- Checking for module 'sdl2>=2.0.4'
--
-- Checking for module 'libavcodec'
--   No package 'libavcodec' found
-- Checking for module 'libavutil'
--   No package 'libavutil' found
-- Checking for module 'x11'
--   No package 'x11' found
-- Checking for module 'vdpau'
--   No package 'vdpau' found
-- Checking for module 'libva'
--   No package 'libva' found
-- Checking for module 'libva-x11'
--   No package 'libva-x11' found
-- Checking for module 'libcec>=4'
--
-- Checking for module 'egl'
--   No package 'egl' found
-- Checking for module 'glesv2'
--   No package 'glesv2' found
CMake Error at CMakeLists.txt:166 (message):
  No video output available

-- Configuring incomplete, errors occurred!
See also "/root/Software/moonlight-embedded/build/CMakeFiles/CMakeOutput.log".
See also "/root/Software/moonlight-embedded/build/CMakeFiles/CMakeError.log".

Sk1pp3r, 2018-08-29:
After installing the dependencies from [compilation wiki](https://github.com/irtimmer/moonlight-embedded/wiki/Compilation), I get the following:
```
rock64@rock64:~/Software/moonlight-embedded/build# cmake ../
-- Could NOT find Broadcom (missing:  BROADCOM_INCLUDE_DIR VCOS_LIBRARY VCHIQ_LIBRARY OPENMAXIL_LIBRARY BCM_HOST_LIBRARY)
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Could NOT find Amlogic (missing:  AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY)
-- Could NOT find Rockchip (missing:  ROCKCHIP_INCLUDE_DIR ROCKCHIP_LIBRARY)
-- Checking for module 'sdl2>=2.0.4'
--
-- Checking for module 'libavcodec'
--   No package 'libavcodec' found
-- Checking for module 'libavutil'
--   No package 'libavutil' found
-- Checking for module 'x11'
--   No package 'x11' found
-- Checking for module 'vdpau'
--   No package 'vdpau' found
-- Checking for module 'libva'
--   No package 'libva' found
-- Checking for module 'libva-x11'
--   No package 'libva-x11' found
-- Checking for module 'libcec>=4'
--
-- Checking for module 'egl'
--   No package 'egl' found
-- Checking for module 'glesv2'
--   No package 'glesv2' found
CMake Error at CMakeLists.txt:166 (message):
  No video output available

-- Configuring incomplete, errors occurred!
See also "/root/Software/moonlight-embedded/build/CMakeFiles/CMakeOutput.log".
See also "/root/Software/moonlight-embedded/build/CMakeFiles/CMakeError.log".
```

irtimmer, 2018-08-29:
_libraspberrypi-dev_ still does seem to be missing, otherwise provide the list of installed files for this package using `dpkg-query -L libraspberrypi-dev`

Sk1pp3r, 2018-09-03:
@irtimmer actually, rbp-userland-osmc and libraspberrypi0 (and their `dev` versions) were not found during `apt-get install`.
it would probably be best to find some kind of alternatives to these packages

Sk1pp3r, 2018-09-16:
@irtimmer any updates on this issue?
I found `libavcodec` as part of  `ffmpeg` library, which i installed, but I still get this part when trying to build:
```
-- Checking for module 'libavcodec'
--   No package 'libavcodec' found
```

DDMP6, 2019-02-05:
this is frustrating to see, I am sure the devs would save themselves a lot of bug reports, if there would be single HOW TO INSTALL properly documented.

maretodoric, 2019-04-21:
Ok, i have to chime in. After thoroughly looking into CMakeLists to see what it's actually looking for and searching through raspbian repositories, I've found that taking following steps works (At least I've managed to compile and install using this method)

Some "missing files" will appear during cmake but it seems that they can be safely ignored since those files are used on some other devices rather than raspberry pi (at least i think).

Ome more thing, this is tested on Raspbarry Pi 3B+, with Raspbian 9 (Stretch)

```
sudo apt install libopus-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libevdev-dev libexpat1-dev libpulse-dev uuid-dev libenet-dev gcc g++ libraspberrypi-dev fakeroot debhelper libx11-dev libvdpau-dev libva-dev libgegl-dev cmake make -y
git clone https://github.com/irtimmer/moonlight-embedded.git
cd moonlight-embedded
git submodule update --init --recursive
mkdir build
cd build
cmake ../
make
sudo make install
sudo ln -s /usr/local/bin/moonlight /usr/bin/moonlight
sudo ldconfig
```

I had to add symlink because for some reason, it was looking for moonligth in /usr/bin, even though PATH is sed for /usr/local/bin as well !
Hope it helps.