moonlight-stream/moonlight-embedded #663: Build fails with latest FFMPEG

ShoelaceMan, 2018-05-17:
Hello,

On the latest version of FFMPEG, moonlight-embedded no longer builds successfully. I tried using the suggested replacements from the compiler output, but it failed still with a different error. Here's compiler output:

```
[  1%] Built target docs
[ 38%] Built target moonlight-common
[ 57%] Built target gamestream
[ 61%] Building C object CMakeFiles/moonlight.dir/src/video/sdl.c.o
[ 63%] Building C object CMakeFiles/moonlight.dir/src/video/x11.c.o
[ 61%] Building C object CMakeFiles/moonlight.dir/src/video/ffmpeg.c.o
[ 65%] Building C object CMakeFiles/moonlight.dir/src/video/ffmpeg_vdpau.c.o
[ 67%] Building C object CMakeFiles/moonlight.dir/src/audio/pulse.c.o
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg.c: In function ‘ffmpeg_init’:
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg.c:55:3: warning: ‘avcodec_register_all’ is deprecated [-Wdeprecated-declarations]
   avcodec_register_all();
   ^~~~~~~~~~~~~~~~~~~~
In file included from /tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg.h:22,
                 from /tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg.c:20:
/usr/include/libavcodec/avcodec.h:4086:6: note: declared here
 void avcodec_register_all(void);
      ^~~~~~~~~~~~~~~~~~~~
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg.c:99:27: error: ‘CODEC_FLAG_LOW_DELAY’ undeclared (first use in this function); did you mean ‘AV_CODEC_FLAG_LOW_DELAY’?
     decoder_ctx->flags |= CODEC_FLAG_LOW_DELAY;
                           ^~~~~~~~~~~~~~~~~~~~
                           AV_CODEC_FLAG_LOW_DELAY
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg.c:99:27: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [CMakeFiles/moonlight.dir/build.make:193: CMakeFiles/moonlight.dir/src/video/ffmpeg.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/x11.c: In function ‘x11_setup’:
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/x11.c:96:48: error: ‘FF_INPUT_BUFFER_PADDING_SIZE’ undeclared (first use in this function); did you mean ‘AV_INPUT_BUFFER_PADDING_SIZE’?
   ffmpeg_buffer = malloc(DECODER_BUFFER_SIZE + FF_INPUT_BUFFER_PADDING_SIZE);
                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                AV_INPUT_BUFFER_PADDING_SIZE
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/x11.c:96:48: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [CMakeFiles/moonlight.dir/build.make:245: CMakeFiles/moonlight.dir/src/video/x11.c.o] Error 1
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c: In function ‘vdp_get_free_render_state’:
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c:59:22: error: dereferencing pointer to incomplete type ‘struct vdpau_render_state’
     if (!render_state->state)
                      ^~
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c:67:59: error: invalid application of ‘sizeof’ to incomplete type ‘struct vdpau_render_state’
   struct vdpau_render_state* render_state = malloc(sizeof(struct vdpau_render_state));
                                                           ^~~~~~
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c:70:34: error: invalid application of ‘sizeof’ to incomplete type ‘struct vdpau_render_state’
   memset(render_state, 0, sizeof(struct vdpau_render_state));
                                  ^~~~~~
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c: In function ‘vdp_get_buffer’:
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c:86:26: error: ‘FF_VDPAU_STATE_USED_FOR_RENDER’ undeclared (first use in this function)
   pRenderState->state |= FF_VDPAU_STATE_USED_FOR_RENDER;
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c:86:26: note: each undeclared identifier is reported only once for each function it appears in
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c: In function ‘vdp_get_format’:
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c:91:10: error: ‘AV_PIX_FMT_VDPAU_H264’ undeclared (first use in this function); did you mean ‘AV_PIX_FMT_VDPAU’?
   return AV_PIX_FMT_VDPAU_H264;
          ^~~~~~~~~~~~~~~~~~~~~
          AV_PIX_FMT_VDPAU
make[2]: *** [CMakeFiles/moonlight.dir/build.make:284: CMakeFiles/moonlight.dir/src/video/ffmpeg_vdpau.c.o] Error 1
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/sdl.c: In function ‘sdl_setup’:
/tmp/makepkg/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/sdl.c:43:48: error: ‘FF_INPUT_BUFFER_PADDING_SIZE’ undeclared (first us
```

**_NVidia Geforce Experience version:_**
Latest

**Moonlight Embedded version:**
Latest

**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
Both AUR, current release, and git clone

**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
Desktop computer

**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._
Arch Linux

**What is the expected result?**
Build succeeds

**What happens instead of that?**
Build fails

mrfixit2001, 2018-05-26:
Just ran into this issue myself. Here is the patch I wrote that you need to fix it.

`--- a/src/video/ffmpeg.c	2018-01-14 09:09:59.000000000 -0500
+++ b/src/video/ffmpeg.c	2018-05-25 20:43:49.296264232 -0400
@@ -96,7 +96,7 @@
 
   if (perf_lvl & LOW_LATENCY_DECODE)
     // Use low delay single threaded encoding
-    decoder_ctx->flags |= CODEC_FLAG_LOW_DELAY;
+    decoder_ctx->flags |= AV_CODEC_FLAG_LOW_DELAY;
 
   if (perf_lvl & SLICE_THREADING)
     decoder_ctx->thread_type = FF_THREAD_SLICE;
--- a/src/video/sdl.c	2018-01-14 09:09:59.000000000 -0500
+++ b/src/video/sdl.c	2018-05-25 20:46:35.792264232 -0400
@@ -40,7 +40,7 @@
     return -1;
   }
   
-  ffmpeg_buffer = malloc(DECODER_BUFFER_SIZE + FF_INPUT_BUFFER_PADDING_SIZE);
+  ffmpeg_buffer = malloc(DECODER_BUFFER_SIZE + AV_INPUT_BUFFER_PADDING_SIZE);
   if (ffmpeg_buffer == NULL) {
     fprintf(stderr, "Not enough memory\n");
     ffmpeg_destroy();`

DocMAX, 2018-05-26:
Doesn't work

```
/home/docmax/.cache/pacaur/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c: In function 'vdp_get_free_render_state':
/home/docmax/.cache/pacaur/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c:59:22: error: dereferencing pointer to incomplete type 'struct vdpau_render_state'
     if (!render_state->state)
                      ^~
/home/docmax/.cache/pacaur/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c:67:59: error: invalid application of 'sizeof' to incomplete type 'struct vdpau_render_state'
   struct vdpau_render_state* render_state = malloc(sizeof(struct vdpau_render_state));
                                                           ^~~~~~
/home/docmax/.cache/pacaur/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c:70:34: error: invalid application of 'sizeof' to incomplete type 'struct vdpau_render_state'
   memset(render_state, 0, sizeof(struct vdpau_render_state));
                                  ^~~~~~
/home/docmax/.cache/pacaur/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c: In function 'vdp_get_buffer':
/home/docmax/.cache/pacaur/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c:86:26: error: 'FF_VDPAU_STATE_USED_FOR_RENDER' undeclared (first use in this function)
   pRenderState->state |= FF_VDPAU_STATE_USED_FOR_RENDER;
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/docmax/.cache/pacaur/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c:86:26: note: each undeclared identifier is reported only once for each function it appears in
/home/docmax/.cache/pacaur/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c: In function 'vdp_get_format':
/home/docmax/.cache/pacaur/moonlight-embedded/src/moonlight-embedded-2.4.6/src/video/ffmpeg_vdpau.c:91:10: error: 'AV_PIX_FMT_VDPAU_H264' undeclared (first use in this function); did you mean 'AV_PIX_FMT_VDPAU'?
   return AV_PIX_FMT_VDPAU_H264;
          ^~~~~~~~~~~~~~~~~~~~~
          AV_PIX_FMT_VDPAU
make[2]: *** [CMakeFiles/moonlight.dir/build.make:297: CMakeFiles/moonlight.dir/src/video/ffmpeg_vdpau.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:69: CMakeFiles/moonlight.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

mrfixit2001, 2018-05-26:
The patch I sent didn't address the two ffmpeg files that are optionally included, such as your vdpau example, because my build didn't require them. Sorry about that! It can be easily updated for those files, as The issue is simply that the new ffmpeg version adds an "AV" before the variables. As a 2 second tip to making patches, find the file it's complaining about and make a copy of it. Then open it in a text editor and make the change you want. Then at command prompt in the directory of those files run "diff -Naur <unedited copy> <edited copy>  > patch" and it makes a file named "patch". That patch file just needs a quick update now to correctly reference the right file names on the @@ lines. See my submitted patch for reference.

DocMAX, 2018-05-26:
There is no "AV" alternative for FF_VDPAU_STATE_USED_FOR_RENDER

mrfixit2001, 2018-05-26:
Looks like they removed FF_VDPAU_STATE_USED_FOR_RENDER and FF_VDPAU_STATE_USED_FOR_REFERENCE completely in October 2017-
https://github.com/FFmpeg/FFmpeg/commit/c68a3ab96ec0497ae2d627ddd30c61737d18173e#diff-166aaf5470fb0f2476822e3fb7d26757

That commit shows their previous values (1 and 2, respectively), so just update the source to use 1 and 2 instead.

DocMAX, 2018-05-31:
the whole thing is messed up! it has to be checked from scratch!

irtimmer, 2018-08-11:
Should be fixed with last release

closed 2018-08-11 by irtimmer