moonlight-stream/moonlight-embedded #666: Bethesda games fail with Transaction failed: 11

ms5991, 2018-06-06:
Please provide the following info.

**_NVidia Geforce Experience version:** 3.14.0.139
**Moonlight Embedded version:** 2.4.6
**Moonlight Embedded source:** Repository 
**Moonlight Embedded running on:** Raspberry Pi 3 B+
**Moonlight Embedded running on distribution:** Raspbian Stretch

**Verbose output `-verbose` of Moonlight Embedded:**
```
Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
Connect to 192.168.113.6...
NVIDIA GeForce GTX 980, GFE 3.14.0.139 (gs_04_06_24080785, 7.1.406.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

```
**What is the expected result?**  Game starts.

**What happens instead of that?**  When attempting to launch Skyrim or Fallout 4, the launcher for the game shows up briefly, then the streaming ends with the above output.  I tried a different game, non-Bethesda without a launcher (Just Cause 2), and it worked fine.  Therefore I wonder if the issue is related specifically to the fact that there is a launcher, or specifically to the fact that both these games are Bethesda (maybe a shared setting that causes the problem).  

Additional details: I have a GTX 980 with the most up to date driver.  Everything is wired via Ethernet through my router and a Netgear Gigabit switch.  

There is a similar error that occurs occasionally with an additional line:

`Input: sendInputPacketOnControlStream() failed: -1`

DaOliver86, 2018-06-11:
You may want to check if your issue is steam related. I wittness the same error, but only with steam games. Extern software added into the steam library works well but starting a steam game brings me back to retropie start screen with your error in the output screen

ms5991, 2018-06-11:
I ended up figuring out a solution, though it seems like a workaround rather than a fix.  If I use `-app 'Fallout 4'` or `-app 'The Elder Scrolls V: Skyrim Special Edition'`, the games launch fine.  Only if I use Steam Big Picture to launch the game do I see the error.  It does seem like a possible Steam issue.  Therefore, I'm satisfied for now, but I'll leave the issue open in case the devs want to investigate.

irtimmer, 2018-08-12:
It seems not something that can be fixed within Moonlight and probably needs a fix in Steam or GFE.

closed 2018-08-12 by irtimmer