moonlight-stream/moonlight-embedded #668: 8bitdo SF30 Pro not recognize as controller

xenofree, 2018-06-16:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.10.0.95
**Moonlight Embedded version:** 2.4.6
**Moonlight Embedded source:** repository
**Moonlight Embedded running on:** Raspberry Pi 3
**Moonlight Embedded running on distribution:** Raspbian

**Verbose output `-verbose` of Moonlight Embedded:**

pi@retropie:~ $ sudo moonlight map -input /dev/input/event0 -verbose
Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
No mapping available for 8Bitdo SF30 Pro (05000000c82d00000061000000010000) on /dev/input/event0
*** buffer overflow detected ***: moonlight terminated
Aborted

I tried to add it to the gamecontrollerdb.txt but it don't work anyway

pi@retropie:~ $ grep 05000000c82d00000061000000010000 /usr/share/moonlight/gamecontrollerdb.txt
05000000c82d00000061000000010000,8Bitdo SF30 Pro,a:b1,b:b0,back:b10,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b2,leftshoulder:b6,leftstick:b13,lefttrigger:a5,leftx:a0,lefty:a1,rightshoulder:b7,rightstick:b14,righttrigger:a4,rightx:a2,righty:a3,start:b11,x:b4,y:b3,

5schatten, 2018-07-06:
I guess you tried to set up the controller in Dinput mode? Didn't work for me too. Once I've set it up in Xinput mode I was able to map the controlls and Steam recognized it however it didn't really worked for me.
The only way I got it working was Xinput mode & cable based connection. Then it works properly however the trigger buttons won't work.

All in all it looks like the Xinput mode is utterly broken and we need a Dinput to Xinput wrapper as workaround.

5schatten, 2018-07-06:
@xenofree 
Start your controller in **Switch** mode (**Start+Y**) and use this config:

`050000007e0500000920000001000000,Pro Controller,a:b0,b:b1,x:b2,y:b3,start:b9,guide:b12,back:b8,leftstick:b10,rightstick:b11,leftshoulder:b4,rightshoulder:b5,dpup:h0.9,dpleft:h0.8,dpdown:h0.12,dpright:h0.2,leftx:a0,lefty:a1,rightx:a2,righty:a3,lefttrigger:b6,righttrigger:b7,`

Works for me.

closed 2018-08-11 by irtimmer