moonlight-stream/moonlight-embedded #669: linux mint compilation

iakiz, 2018-06-17:
I have a problem on step 3 cmake ../ from wiki manual compilation.

-- Could NOT find ALSA (missing:  ALSA_LIBRARY ALSA_INCLUDE_DIR) 
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Opus (missing: OPUS_LIBRARY OPUS_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindOpus.cmake:59 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:13 (find_package)

All needed packages installed:
 
libopus0
libexpat1
libasound2
libudev0 or libudev1
libavahi-client3
libcurl3
libevdev2
libenet7
cmake

My system Linux Mint 18.3 Sylvia

Sk1pp3r, 2018-06-19:
I have the similar issue only on a different device #654 
This keeps driving me nutts.

irtimmer, 2018-08-12:
Duplicate of #654

closed 2018-08-12 by irtimmer