moonlight-stream/moonlight-embedded #671: Video Stream Start Failed: -2

MichaelAOlson, 2018-06-27:
**_NVidia Geforce Experience version:_** 3.14.0.139
**Moonlight Embedded version:** 2.4.6
**Moonlight Embedded source:** included in distribution
**Moonlight Embedded running on:** Raspberry Pi 3 B
**Moonlight Embedded running on distribution:** Raspbian 9.4 "Stretch"

**Verbose output `-verbose` of Moonlight Embedded:**
Connect to 192.168.1.11...
NVIDIA GeForce GTX 770, GFE 3.14.0.139 (gs_04_06_24080785, 7.1.406.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Detected Afterglow Gamepad for Xbox 360 (030000006f0e00001302000000010000) on /dev/input/event1 as Xbox Gamepad (userspace driver)
Stream 1920x1080, 60fps, 20000kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Can't setup video
Starting video stream...Video stream start failed: -2
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
moonlight stream -fps 60 -1080 -app Steam 192.168.1.11
Also doesn't work for any other fps/resolution combination

**What happens instead of that?**
Fails to start video, Raspberry Pi becomes entirely unresponsive.

MichaelAOlson, 2018-06-27:
The source code line where it fails is here: 
https://github.com/irtimmer/moonlight-embedded/blob/9b0b2d4a6e06398fc9b164dc53e96455314a8dd1/src/video/pi.c#L159
I plan to look into what this if-statement is checking for tomorrow morning.

MichaelAOlson, 2018-06-27:
When building from source rather than from the package supplied in
`deb http://archive.itimmer.nl/raspbian/moonlight stretch main`
I get a black screen, which is also unresponsive. Is there a compatibility list with GeForce Experience versions and GeForce Game Ready driver versions?

cgutman, 2018-06-28:
There haven't been breaking changes in the GameStream protocol for several months. Master should be good through GFE 3.14.

closed 2019-04-30 by irtimmer