moonlight-stream/moonlight-embedded #673: Suddenly Sound is extremely slowed and weired. Video is fine.

Spamm00r, 2018-07-05:
Please provide the following info.

**_NVidia Geforce Experience version: 385,69 GTX 960 Ti NVIDIA GeForce GTX 960, GFE 3.9.0.61 (gs_04_00_22663502, 7.1.400.0)
**Moonlight Embedded version:latest. Updated yesterday via respository. Moonlight Embedded 2.4.3 (EMBEDDED;ALSA;PI)
**Moonlight Embedded source:Updated yesterday via respository.
Moonlight Embedded running on: Raspberry Pi 1
**Moonlight Embedded running on distribution:OSMC

Verbose output -verbose of Moonlight Embedded: ? 

**What is the expected result? Moonlight viodeo and sound schould be fine.

Recently the sound on my moonlight session is very slowed down and weird, but the Video streams just fine. 

Moonlight session on my Iphone app work fine. issue only appears on Raspebrry moonlight session. It used to work fine for a long time, only recently the sound issue started. A reinstall did not help.

What might be the issue?

Spamm00r, 2018-07-11:
Went back to nvidia driver from January 2018 and switched sound device in luna settings and everything is back to Normal.

NEVER UPDATE NVIDIA drivers. Moonlight always, always breaks after Updates.

Spamm00r, 2018-09-11:
There and Back again!

Unfortunately Geforce Experience updated itself and thats when the trouble started again.  I thought I new what I ahd to do, but I downgraded to 4 different Geforce Experience Versions and to older Nvidia drivers as well, but the sound issues remain. I also went back to older moonlight builds, but nothing helps. 

I suspect that the problem lies with Raspberry's sound device. I listen to the sound via the Headphone jack and have the best sound if I select "Analog" in Kodi's Audio Options.

In Luna Launcher however I can not switch the device. It remains on the hdmi device. When the stream starts, I can hear sound but the sound is slowed and distorted. Video however plays fine.

Here is the verbose log:

osmc@osmc:~$ moonlight stream -verbose -app RD x > moonlightverbose.txt
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (lefttrigger:+a3)
Can't map (righttrigger:-a3)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
EVIOCGRAB failed with error 16
EVIOCGRAB failed with error 16

Spamm00r, 2018-09-11:
Moonlight Embedded 2.4.7 (PI;ALSA;EMBEDDED)
NVIDIA GeForce GTX 960, GFE 3.12.0.84 (gs_04_04_23287129, 7.1.404.0)
Platform Raspberry Pi (Broadcom)

Spamm00r, 2018-09-14:
I cant edit my OP:

I now let the stream run for several minutes and the "Received OOS audio data" gets fewer over time so that at some point the sound is comprehenisble and only occassionally the sound glitches due to a "OOS audio data".

This behaviour is reproducable. It always starts out with many "Received OOS audio data" and only after a few minutes, the Received OOS audio data frequency goes down.

I could rule out that ther eis a network issue. My network connection is sufficient and the streaming on my iphone app or another windows pc via Moonlight windows works flawless. 

So the problem just occurs when i try to stream with moonlight embedded.

I'm right now on the latest version of Nvidia/GFE and Moonlight. Cahnging nvidia/GFE versions to as far back as 2016 did not make any difference.

kurac338, 2018-12-18:
Have you ever had it working properly on the Pi 1?
Personally I would say that the Pi 1 is too weak for the things you are asking from it. OSMC + Moonlight running in parallel. I have it on a Pi 3 (also with OSMC and Luna) without issues.
It could be that your Pi is experiencing a race condition and overloading the CPU so it starts to drop audio in favor of the video stream.