moonlight-stream/moonlight-embedded #675: Wrong mapping for Select Button on Xbox One Controller

Fabiosilvero, 2018-07-22:
Please provide the following info.

**_NVidia Geforce Experience version:_** GFE 3.12.0.84
**Moonlight Embedded version:**Moonlight Embedded 2.4.6
**Moonlight Embedded source:** From deb http://archive.itimmer.nl/raspbian/moonlight stretch main
**Moonlight Embedded running on:** _Raspberry Pi
**Moonlight Embedded running on distribution:**Raspbian Stretch Up to date

**Verbose output `-verbose` of Moonlight Embedded:**

$ ml -verbose
Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to dest_host
NVIDIA GeForce GTX 1050 Ti, GFE 3.12.0.84 (gs_04_04_23287129, 7.1.404.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Detected Microsoft Corporation Gamepad (050000005e040000fd02000003090000) on /dev/input/event0 as Xbox One Wireless Controller
Detected PLAYSTATION(R)3 Controller (060000004c0500006802000000010000) on /dev/input/event2 as PS3 Controller (Bluetooth)
Mapping for PLAYSTATION(R)3 Controller (060000004c0500006802000000010000) on /dev/input/event2 is incorrect
Stream 1920 x 1080, 30 fps, 12000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Stopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed
Loss Stats: Transaction failed: 11
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
Select button for Xbox One Controller working as a Select Button

**What happens instead of that?**
Select Button act as a D-Hat Up Button in every game (CEMU, FFXV...) 

Others buttons work fine tho.

Here's the mapping for the Xbox One Controller from gamecontroller.db : 

050000005e040000fd02000003090000,Xbox One Wireless Controller,a:b0,b:b1,back:b15,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dp
up:h0.1,guide:b16,leftshoulder:b6,leftstick:b13,lefttrigger:a5,leftx:a0,lefty:a1,rightshoulder:b7,rightstick:b14,righttrigger:a4,rightx:a2,righty:a3,start:b11,x:b3,y:b4,

ziotidus, 2018-08-02:
Hi i've the same problem with the PS3 controller:

Spa1ner, 2018-08-03:
Also I have the same problem, 

- Mapping for PLAYSTATION(R)3 Controller (060000004c0500006802000000010000) on /dev/input/event2 is incorrect

Should be a guide how to, there are so many problems with the controllers. I dont know what I have to do.

pokemonw2, 2018-08-20:
Anyone found a solution to this issue? 
I also have a xbox one s wireless controller which moonlight doesnt interface well with the home and select button. Both of them register as an "up" in steam big picture.

TorarinnGarlick, 2018-10-17:
Same issue with my xbox one s controller on my RPI3(retropie).

TheUntouchable, 2018-10-20:
Same problem here with RPI3 / Raspbian and XBOX One Controller (Bluetooth). The old XBOX Controller with the dongle is working with the xboxdrv without any problems..

commit 4b41692f (2021-08-08): Fix handling of buttons outside the normal joystick range

See https://github.com/libsdl-org/SDL/commit/aa03b9d7afedefddda4e7f1575d87865ee5f3d93

Fixes #788
Fixes #696
Fixes #675
Fixes #693
Fixes #642
closed 2021-08-08 by cgutman