moonlight-stream/moonlight-embedded #676: Mapping is incorrect

BrutolocoW, 2018-07-23:
**_NVidia Geforce Experience version:_** 
3.14.1.48
**Moonlight Embedded version:** 
2.4.6
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
http://archive.itimmer.nl/raspbian/moonlight jessie main
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
Raspberry Pi 2 Model B Revision 2.0 256MB (Qisda)

**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._
Raspbian GNU/Linux 8 (Jessie)

**Verbose output `-verbose` of Moonlight Embedded:**

pi@raspberrypi:~ $ moonlight stream -1080 -app mstsc -verbose
Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to 192.168.1.123...
NVIDIA GeForce GTX 750 Ti, GFE 3.14.1.48 (gs_04_08_24506279, 7.1.408.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Detected ShanWan     PC/PS3/Android Gamepad (03000000632500007505000011010000) on /dev/input/event2 as ShanWan PC/PS3/Android Gamepad
Mapping for ShanWan     PC/PS3/Android Gamepad (03000000632500007505000011010000) on /dev/input/event2 is incorrect
Stream 1920 x 1080, 30 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Stopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed
Loss Stats: Transaction failed: 11
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**

The gamepad should be mapped because I edited correctly /usr/share/moonlight/gamecontrollerdb.txt as I read in another issues here adding my gamepad. 
I used sdl-jstest/jstest-sdl/gamepadtool for test and accurate all keys bindings, like I've read in another issues.

**What happens instead of that?**

"Mapping is incorrect"

-----------------------------------------------------------

I don't know what to do now. What I'm doing wrong.

This is the gamepad I'm using 
https://www.gearbest.com/game-controllers/pp_424407.html
https://www.amazon.com/CFStore-Bluetooth-Controller-Rechargeable-GAME-Btooth-BK/dp/B06XB6PQSR/ref=sr_1_2?ie=UTF8&qid=1532318407&sr=8-2&keywords=gen+game

have many types of configurations for being connected. I'm trying connecting it usb cable connection-DirectInput mode.

I tried putting putting hat as axis, like is show in evtest command, and fail too. This is what evtest command shows:

-------------------------------
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x2563 product 0x575 version 0x111
Input device name: "ShanWan     PC/PS3/Android Gamepad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 306 (BTN_C)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 309 (BTN_Z)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 312 (BTN_TL2)
    Event code 313 (BTN_TR2)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    128
      Min        0
      Max      255
      Flat      15
    Event code 1 (ABS_Y)
      Value    128
      Min        0
      Max      255
      Flat      15
    Event code 2 (ABS_Z)
      Value    128
      Min        0
      Max      255
      Flat      15
    Event code 5 (ABS_RZ)
      Value    128
      Min        0
      Max      255
      Flat      15
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
-------------------------------------

All inputs are tested, detected and work.

This is my gamepad config string, its supposed to be exactly what its should to be each key, tested with gamepadtool in linux, jstest-sdl and sdl-jstest:

03000000632500007505000011010000,ShanWan PC/PS3/Android Gamepad,a:b2,b:b1,x:b3,y:b0,back:b8,guide:b12,start:b9,leftstick:b10,rightstick:b11,leftshoulder:b4,rightshoulder:b5,dpup:h0.1,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,leftx:a0,lefty:a1,rightx:a2,righty:a3,lefttrigger:b6,righttrigger:b7,

Always "mapping is incorrect", and doesn't work

Any clue?

Spa1ner, 2018-08-03:
Same problem here, with PS3 Controller, I think we wont find the fix, I will buy a steamlink, plug and play. It's so annoying, 1 week and nothing I do fix this error. In RetroPie works fine.

Spa1ner, 2018-08-08:
Try this #679

irtimmer, 2018-08-12:
Duplicate of #679

closed 2018-08-12 by irtimmer