moonlight-stream/moonlight-embedded #679: Issue with controllers (xbox1S and dualshock4) not working

BlueGeckoO, 2018-08-03:
Please provide the following info.

**_NVidia Geforce Experience version:_** _GFE 3.14.1.48_
**Moonlight Embedded version:** _2.4.6_
**Moonlight Embedded source:** _repository_
**Moonlight Embedded running on:** _Raspberry Pi 3B+_
**Moonlight Embedded running on distribution:** _Raspbian_

**Verbose output `-verbose` of Moonlight Embedded:**
$moonlight stream -verbose -mapping /usr/share/moonlight/gamecontrollerdb.txt -input /dev/input/event0 192.168.1.10
Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
Connect to 192.168.1.10...
NVIDIA GeForce GTX 1080 Ti, GFE 3.14.1.48 (gs_04_08_24506279, 7.1.408.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Add input /dev/input/event0...
Detected Microsoft X-Box One S pad (030000005e040000ea02000001030000) on /dev/input/event0 as Microsoft X-Box One pad v2
Stream 1280x720, 60fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done

**What is the expected result?**

Being able to play with the controller

**What happens instead of that?**

Steam big picture starts but unable to navigate through it with the controller, as sticks, buttons and triggers seem to do nothing. Only mouse/keyboard works if I don't add the "-input /dev/input/event0" option.

I also tried to start directly a few games instead of Steam but same issue: the game starts but I can't do anything with a controller.

Finally I tried a PS4 Controller but the result is the same.

Both Xbox 1S and PS4 controllers works with other softwares on my rapsberry pi (raspbian stretch) like emulationstation, and Steam In-home streaming works from another PC client. 

Any help is appreciated and I stay available for questioning :-)

dmarszal, 2018-08-04:
Confirming this, worked around 2 weeks ago, tried playing today and pad doesn't work.
Input is detected while creating custom mapping. Looks like GeForce Gamestream is ignoring the input.

Release: 2.4.6
System: Ubuntu 18.04
GeForce Experience: 3.14.1.48

Starting moonlight with:
$ moonlight stream 192.168.1.xxx -app "Stream" -debug

Output lists:

Detected Microsoft X-Box One S pad (030000005e040000ea02000001030000) on /dev/input/event4 as Microsoft X-Box One pad v2

Nothing changed on the hardware side.

teesee23, 2018-08-05:
I had exactly the same issue using an 8bitdo SN30pro gamepad and Moonlight on my retropie. Gamepad was working, then stopped a few weeks ago.

I was using the moonlight version installed with the script found here: https://github.com/TechWizTime/moonlight-retropie
which pulls from the repo at deb http://archive.itimmer.nl/raspbian/moonlight jessie main

I fixed the issue by building moonlight-embedded from source using the instructions provided here:
https://github.com/irtimmer/moonlight-embedded/wiki/Compilation

Seems to work fine for me now!

@irtimmer I guess your repo needs updating? Anyhow, thanks for maintaining a great project, by far the most used software on my raspberry pi!

BlueGeckoO, 2018-08-05:
Thanks @teesee23, I'll try compilation to see if it fix my problem too!

dmarszal, 2018-08-06:
Thanks for the feedback. Can you please explain which version did you compile? Current master branch maybe (https://github.com/irtimmer/moonlight-embedded/tree/master)? 
My first feedback was for self-compiled https://github.com/irtimmer/moonlight-embedded/releases/tag/v2.4.6 and it's not working there.

teesee23, 2018-08-06:
@dmarszal correct, I built the current master branch.

Good luck!

BlueGeckoO, 2018-08-06:
I just did a 
`git clone https://github.com/irtimmer/moonlight-embedded.wiki.git`

I guess it the master branch ..

controlstudent, 2018-08-08:
Hi @teesee23, I am having the exact same issue as OP.  

> I fixed the issue by building moonlight-embedded from source using the instructions provided here:
https://github.com/irtimmer/moonlight-embedded/wiki/Compilation

Can you explain how to do this explicitly? I am not strong in RPi/Linux script in general.

Thank you,

dmarszal, 2018-08-08:
Is libgamestream compiled?
make install should print libgamestream lines as below.
$ sudo make install                                                                                                                                                                                                  [sudo] password for darek:                                                                                                                                                                                                                                              [ 37%] Built target moonlight-common                                                                                                                                                                                                                                    [ 56%] Built target gamestream                                                                                                                                                                                                                                          [ 98%] Built target moonlight                                                                                                                                                                                                                                           [100%] Built target docs                                                                                                                                                                                                                                                Install the project...                                                                                                                                                                                                                                                  -- Install configuration: ""                                                                                                                                                                                                                                            -- Installing: /usr/local/bin/moonlight                                                                                                                                                                                                                                 -- Set runtime path of "/usr/local/bin/moonlight" to ""                                                                                                                                                                                                                 -- Up-to-date: /usr/local/share/moonlight/gamecontrollerdb.txt                                                                                                                                                                                                          -- Up-to-date: /usr/local/etc/moonlight.conf                                                                                                                                                                                                                            -- Installing: /usr/local/lib/libgamestream.so.2.4.6                                                                                                                                                                                                                    -- Up-to-date: /usr/local/lib/libgamestream.so.4                                                                                                                                                                                                                        -- Up-to-date: /usr/local/lib/libgamestream.so                                                                                                                                                                                                                          -- Set runtime path of "/usr/local/lib/libgamestream.so.2.4.6" to ""                                                                                                                                                                                                    -- Installing: /usr/local/lib/libmoonlight-common.so.2.4.6                                                                                                                                                                                                              -- Up-to-date: /usr/local/lib/libmoonlight-common.so.4                                                                                                                                                                                                                  -- Up-to-date: /usr/local/lib/libmoonlight-common.so                                                                                                                                                                                                                    -- Installing: /usr/local/share/man/man1/moonlight.1               

If they are there check if /usr/local/lib is in ld.so.conf.d

$ grep -R "/usr/local/lib/" /etc/ld.so.conf.d/
I had to create one liner /etc/ld.so.conf.d/moonlight.conf 
$ cat /etc/ld.so.conf.d/moonlight.conf                                                                                                                                                                               /usr/local/lib/                 

After that run
$ sudo ldconfig

Spa1ner, 2018-08-08:
Fixed too!!

controlstudent, 2018-08-10:
@dmarszal Hey, when I try this what you suggested, I get an error I dont know how to deal with.

` $ sudo make install
make: *** No rule to make target 'install'.  Stop.
`

This seems extremely rudimentary, I apologize in advanced. Can you help me out?

controlstudent, 2018-08-10:
@JohnyJack01 If you have success, do you mind sharing what scripts you executed?

cgutman, 2018-08-11:
The 2.4.7 release should fix this issue.